luxonis/depthai-ros #138: publish issue in ubuntu 20.04 and ros2 galactic

JIMyungSik, 2022-09-13:
I am testing my oak-d camera with depthai_examples.
I tried to run with ```ros2 launch depthai_examples rgb_stereo_node.launch.py```.
When run hz command (```ros2 topic hz /color/video/image```) or echo command(```ros2 topic echo /color/video/image```), it doesn't subscribed.
Sometimes, it subscribed after running rviz2 or other subscribing node. And it stopped publishing aperiodic.
Does it publish when there is subscriber? or are there any solutions about this issue?

environment
os : ubuntu 20.04
framework : ros2 galactic
camera : oak-d pro
program : depthai_examples rgb_stereo_node

saching13, 2022-09-13:
Hello,
Sorry for the issue. Can you run it with `DEPTHAI_LEVEL=debug ros2 launch depthai_examples rgb_stereo_node.launch.py`  and share the logs.  This more likely looks like a communication error in the ROS. Does this happen only with this package ?

JIMyungSik, 2022-09-15:
[test.log](https://github.com/luxonis/depthai-ros/files/9572375/test.log)
This is log for ```DEPTHAI_LEVEL=debug ros2 launch depthai_examples rgb_stereo_node.launch.py```.
It happens only with this package. I've experienced a communication issue, but it's the first time that it's not published at all, and it's the same even if it's tested on multiple PCs.

closed 2022-09-16 by JIMyungSik