luxonis/depthai-ros #140: ros2 launch depthai_examples yolov4_publisher.launch.py not publishing topics(yolo4_Spatial_detection.txt)

Tanjim13, 2022-09-21:
Hi, I am using turtlebot4. It is not publishing the topics and cannot see the bounding box in rviz2. All other launch files in depthai_examples is working. But yolov4 is not working properly. Should I need to include any dependencies? Cause it is not showing any errors.

saching13, 2022-09-21:
We have not written a plugin for visualization. Try subscribing from command line to cross check.

Tanjim13, 2022-09-21:
@saching13 , I have used ros2 topic echo  /color/yolo4_Spatial_detection and findout this. But the problem is is_tracking: false all the time. If you have not written a plugin for visualization, that means I cannot see the bounding box? right? Also I can  visualize "stereo/depth" and "color/image" topics from rviz2 but without a bounding box. And color/yolov4_Spatial_detections is not visible. 
[yolov4.txt](https://github.com/luxonis/depthai-ros/files/9612499/yolov4.txt)

Tanjim13, 2022-09-21:
Today, I run yolov4, it is not publishing any topics. I am not sure what is the reason. Other scripts in depthai_examples are working properly. But this script is causing the problem.

saching13, 2022-10-03:
@Tanjim13  I have the marker publishing for ROS1 [here](https://github.com/luxonis/depthai-ros/tree/ros-release/depthai_examples/ros1_scripts).

Currently we are swamped. Will add one to ROS2 soon. 
If you write one. Feel free to create a PR.

Serafadam, 2023-06-21:
Added in https://github.com/luxonis/depthai-ros/pull/336

jiangwenjie1011, 2023-07-18:
I found it was because the current provided by the raspberry pie usb interface is too small, I removed the radar usb interface, it works normally now