luxonis/depthai-ros #142: [Feature-Request] {get pointcloud msg}

an99990, 2022-09-24:
### Start with the `why`:
It seems that the only way to get pointcloud if by launching the stereo_intertial_nodeleter file. However, i dont want to have yolov4 running at the same time. My opencv version is 3.4.9, i cant upgrade it because it seems that rtabmap needs something 3.X

### Move to the `what`:
is there a way to deactivate the object detection part when launching the file ?
thank you

saching13, 2022-09-24:
Please check the launch file. There is a argument to disable the detection.

an99990, 2022-09-24:
so what should one do if they want the pointcloud message ? it is not provided in the stereo_node launch file

an99990, 2022-09-24:
hi thank you @saching13 , i am new to ros, could you show me how i would disable it ?

an99990, 2022-09-24:
thank you @saching13 i see it

closed 2022-09-24 by an99990