luxonis/depthai-ros #143: Unable to suscribe the Left and Right image topics in the stereo callback of MSCKF-VIO

ahar, 2022-09-27:
Hi,

I am using OAK-D POE for stereo applications. Although, I can subscribe to the Left and the Right image topics by playing `stereo_node.launch`, when I use [MSCKF-VIO](https://github.com/KumarRobotics/msckf_vio/blob/master/src/image_processor.cpp#L209), the program does not enter inside the stereocallback. Replacing this camera with Intel D415, I checked that the program enters the same callback. What I realize is that (a) `outImageMsg.encoding == "16UC1"` is true for the images obtained from OAK-D POE, whereas the images obtained from the D415 are set `mono8`. However, I naively tried to set the encoding type mono8 by making `is_bigendian=false` in both the cases of [this](https://github.com/luxonis/depthai-ros/blob/ros-release/depthai_bridge/src/ImageConverter.cpp#L130) decision loop.
The image below shows the same. 
[image]
However, I think as the data encoding could not be changed (possibly still), the stereo subscriber does not work.
Whereas, for D415, the corresponding image is shown below, where the program enters in the callback.
[image]

Queries:
1. Is it the right way to analyze the problem? 
2. If so, how to set the image encoding type appropriately to resolve this problem?

Best,

SinnedHog, 2022-09-30:
Hi, from my understanding of MSCKF-VIO code, the stereocallback requires both cam0 and cam1 images to have the exact same timestamp. However, the left and right images from oak-d are not exactly the same (maybe sub micro/micro second difference) and hence why it did not enter the callback function. This small time difference shouldn't affect the performance of the VIO. Maybe you can try to change the sync policy in msckf stereocallback to approximate time policy (http://wiki.ros.org/message_filters) and try again.

saching13, 2022-10-10:
Thanks @SinnedHog .
@ahar  as of `outImageMsg.encoding == "16UC1"` this is true if you are subscribing to disparity or depth. Use `stereo_inertial_node` and enable rectified Left and right by disabling the `depth_aligned`. It should help along with the Suggestion @SinnedHog mentioned for time issue.

Serafadam, 2022-11-09:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.