luxonis/depthai-ros #144: [BUG] {Title of the bug}

an99990, 2022-09-29:
**Check if issue already exists**

topic stereo image is weird, see below screenshot for more details.

**Describe the bug**
I wanted to visualize the topic stereo/image but theres nothing to see.

**Minimal Reproducible Example**
roslaunch depthai_examples stereo_inertial_publisher.launch

**Screenshots**
[image]

**Pipeline Graph**

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - Which OS/OS version are you using? Linux
 - Which ROS version are you using?  ROS1 Melodic

**Additional context**
Add any other context about the problem here.

saching13, 2022-10-03:
@an99990  I don't see anything corrupted above.

saching13, 2022-10-03:
If your issue is same as [here](https://github.com/luxonis/depthai-ros/issues/147). Please follow there. And close this as duplicate.

Huynh-Anh, 2022-10-04:
no its not the same topic. The black Image is the stereo topic and the clear image is another topic

akshat-v, 2022-10-04:
Also facing the same issue.

saching13, 2022-10-10:
One is in millimeter and the other is in meters.

[This](https://github.com/luxonis/depthai-ros/blob/9ac79711a7ae247588a86c75911ddb306e03617f/depthai_examples/launch/stereo_inertial_node.launch#L120-L125) nodelet takes the `/stereo_inertial_publisher/stereo/depth` which is in `mm`  and converts it to `/stereo_inertial_publisher/stereo/image` which is in meters

closed 2022-11-03 by an99990