luxonis/depthai-ros #146: [Feature-Request] Object tracker ROS node

daniqsilva25, 2022-09-30:
I am trying to create a custom converter to publish object tracking data through ROS2. I wrote the code, it compiled successfully, the "tracklets" topic appears while running the node, but nothing is published in the topic.

I have followed the implementation of the [spatial detection converter](https://github.com/luxonis/depthai-ros/blob/ros-release/depthai_bridge/src/SpatialDetectionConverter.cpp) and the [object tracker tutorial on RGB](https://docs.luxonis.com/projects/api/en/latest/samples/ObjectTracker/object_tracker/#object-tracker-on-rgb).

saching13, 2022-10-03:
Can you share your code on how you have created it so that I can assist ?

daniqsilva25, 2022-10-07:
Yes, of course!

I added _DetectionTrackerConverter.cpp_ and _DetectionTrackerConverter.hpp_ to __depthai_bridge__, and then I made the ROS2 node _ssdmobilenet_tracker_allsensors_bridged.cpp_. Here is a [link](https://drive.google.com/drive/folders/1NBD4Fbxlacl0J91xXiJEsrXiArVHqdtq?usp=sharing) to access the code.

Thanks in advance!

saching13, 2022-10-10:
@daniqsilva25  feel free to add the tracker option into the existing `Detection` converter and `SpatialDection` converter.

daniqsilva25, 2022-10-12:
Sorry @saching13, I didn't understand what you mean in "add the tracker option".