luxonis/depthai-ros #147: [BUG] depth image is corrupted

Huynh-Anh, 2022-10-03:
Im trying to visualize the depth information of the oak-d camera but there nothing. Also i would like to understand its shape.
is get (720 ,1280,2), why is there two channels in the depth information?

**Minimal Reproducible Example**
Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)

`roslaunch depthai_examples stereo_inertial_publisher.launch
`
**Attach system log**
[image]
**Additional context**

ros : melodic

saching13, 2022-10-03:
Hello. How are you interpreting it. and what are you trying to interpret it ? 
it is (720, 1280) of type `uint16` since depth is in mm here. And lower the values the blacker the image looks. it means the depth is closer and hard to see it that way. Please check the point cloud visualization that comes with it.

Huynh-Anh, 2022-10-04:
when i checked it said the depth type was uint8 and the shape was (720,1280). Here how i read it
since i cant used cv_bridge, this is how i decode the depth image from topic : `/stereo_inertial_publisher/stereo/depth`

`depth_image = np.frombuffer(depth_msg.data, dtype=np.uint8).reshape(depth_msg.height, depth_msg.width, -1)`

Huynh-Anh, 2022-10-04:
ok so i changed the type to unit16 and it is now only 1channel

closed 2022-10-04 by Huynh-Anh