luxonis/depthai-ros #149: [BUG] Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)

Huynh-Anh, 2022-10-05:
**Describe the bug**
Seen similair bugs but solution did not work for me. This is strange because it was working normally two days ago and now it doesnt..

`roslaunch depthai_examples stereo_intertial_node.launch`

**Screenshots**
```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://192.168.1.154:41717/
SUMMARY
========
PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/detectionClassesCount: 80
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/expTime: 20000
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/manualExposure: False
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId:
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/previewHeight: 416
 * /stereo_inertial_publisher/previewWidth: 416
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/ubuntu/dai_...
 * /stereo_inertial_publisher/rgbResolution: 1080p
 * /stereo_inertial_publisher/rgbScaleDinominator: 3
 * /stereo_inertial_publisher/rgbScaleNumerator: 2
 * /stereo_inertial_publisher/sensIso: 800
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False
NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)
auto-starting new master
process[master]: started with pid [4398]
ROS_MASTER_URI=http://localhost:11311/
setting /run_id to c24da60c-44af-11ed-a0f3-dca632d50a38
process[rosout-1]: started with pid [4409]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [4416]
process[stereo_inertial_publisher-3]: started with pid [4417]
process[nodelet_manager-4]: started with pid [4419]
process[depth_image_convertion_nodelet-5]: started with pid [4424]
process[depth_image_to_rgb_pointcloud-6]: started with pid [4425]
[... 89 lines cut ...]
[ INFO] [1664975722.935901732]: Initializing nodelet with 4 worker threads.
 nnPath ,, /home/ubuntu/dai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
[1280 720 1280](tel:12807201280) 720
Listing available devices...
Device Mx ID: 19443010B100DC1200
[19443010B100DC1200] [1.1.2] [11.064] [SpatialDetectionNetwork(9)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: SUPER
terminate called after throwing an instance of 'std::runtime_error'
  what():  Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 4417, exit code -6, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/c24da60c-44af-11ed-a0f3-dca632d50a38/stereo_inertial_publisher-3.log].
log file: /home/ubuntu/.ros/log/c24da60c-44af-11ed-a0f3-dca632d50a38/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Huynh-Anh, 2022-10-06:
it happens after maybe 30 seconds ? I changed the cable and got a similar error with 
`Original message 'Couldn't read data from stream: 'preview' (X_LINK_ERROR)`

the strange thing is that only happens with the stereo_inertial_node, the other nodes are working ..

saching13, 2022-10-10:
@Huynh-Anh  can you check by disabling the Spatial Detection from the launch file and see if it works ?
And also which depthai-core is installed here?

Serafadam, 2022-11-09:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-09 by Serafadam