luxonis/depthai-ros #150: depthai-ros包编译成功，运行ros2 launch depthai_examples rgb_stereo_none_node.launch.py报错

hukangli, 2022-10-07:
[INFO] [launch]: All log files can be found below /home/ahpc/.ros/log/2022-10-07-11-08-44-728548-lotus-NUC11PAHi7-27710
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgb_stereo_node-1]: process started with pid [27714]
[rgb_stereo_node-1] /home/ahpc/dai_ws/install/depthai_examples/lib/depthai_examples/rgb_stereo_node: symbol lookup error: /home/ahpc/dai_ws/install/depthai_examples/lib/depthai_examples/rgb_stereo_node: undefined symbol: _ZNK15image_transport9Publisher7publishERKN11sensor_msgs3msg6Image_ISaIvEEE
[ERROR] [rgb_stereo_node-1]: process has died [pid 27714, exit code 127, cmd '/home/ahpc/dai_ws/install/depthai_examples/lib/depthai_examples/rgb_stereo_node --ros-args --params-file /tmp/launch_params_x02ni8y1 --params-file /tmp/launch_params_jz2rb1ym --params-file /tmp/launch_params_yy0_z5fs --params-file /tmp/launch_params_wgy4k9o5 --params-file /tmp/launch_params_8mtvcc7k --params-file /tmp/launch_params_jdb09sbr --params-file /tmp/launch_params_no8t3gck --params-file /tmp/launch_params_mbkq265l --params-file /tmp/launch_params_n_n5tr5w --params-file /tmp/launch_params_zfctvdtx --params-file /tmp/launch_params_zihsk2tx --params-file /tmp/launch_params_0i8n6gp2 --params-file /tmp/launch_params_95r4_8sm --params-file /tmp/launch_params_4x302nsk'].

saching13, 2022-10-10:
@hukangli  Can you provide more details about the setup ? And can you check the same with `stereo_inertial_node` ?

hukangli, 2022-10-12:
Sorry to answer for so long，I have solved this problem. The reason is caused by ros environment. I installed ros1、ros2.

closed 2022-10-12 by hukangli