luxonis/depthai-ros #151: [BUG] undefined symbol lookup error at the function - getCameraIntrinsics(CameraBoardSocket, Point2f)

akshay-30, 2022-10-10:
I was trying to run a pipeline with imu, rgb, stereo and depth enabled. I created a left and right image converters using the classes provided in dai namespace. I called the readCalibration() function from the device pointer and calling the calibrationToCameraInfo from the left & right converter nodes. I think here is where the getCameraIntrinsics() is being called wherein the symbol lookup error is popping up.  

Sharing the code I used - 
```
{
  plugin_name_ = plugin_name;
  node_ = node;

  RCLCPP_INFO(
    node_->get_logger(), "Initialising Multiple Camera Interface Plugin for '%s'",
    plugin_name_.c_str());

  loadParams();

  configureDevice(camera_params_.mx_id);

  RCLCPP_INFO(
    node_->get_logger(), "%s device for camera '%s' with MXID : %s, is configured!",
    camera_params_.device_name.c_str(), plugin_name_.c_str(), camera_params_.mx_id.c_str());

  // Create data queue objects
  auto left_queue = device_->getOutputQueue("left", 30, false);
  auto right_queue = device_->getOutputQueue("right", 30, false);

  auto calibration_handler = device_->readCalibration();
  // Add Left Camera Publisher
  left_converter_ = std::make_unique<dai::rosBridge::ImageConverter>(
    plugin_name_ + "_left_camera_optical_frame", true);
  auto left_camera_info = left_converter_->calibrationToCameraInfo(
    calibration_handler, dai::CameraBoardSocket::LEFT,
    camera_params_.resolution_map[camera_params_.mono_resolution][0],
    camera_params_.resolution_map[camera_params_.mono_resolution][1]);

  left_publisher_ =
    std::make_unique<dai::rosBridge::BridgePublisher<sensor_msgs::msg::Image, dai::ImgFrame>>(
      left_queue, node_, plugin_name_ + std::string("/left/image"),
      std::bind(
        &dai::rosBridge::ImageConverter::toRosMsg, left_converter_.get(), std::placeholders::_1,
        std::placeholders::_2),
      kQueueSize, left_camera_info, plugin_name_ + "/left");

  // Add Right Camera Publisher
  right_converter_ = std::make_unique<dai::rosBridge::ImageConverter>(
    plugin_name_ + "_right_camera_optical_frame", true);
  auto right_camera_info = right_converter_->calibrationToCameraInfo(
    calibration_handler, dai::CameraBoardSocket::RIGHT,
    camera_params_.resolution_map[camera_params_.mono_resolution][0],
    camera_params_.resolution_map[camera_params_.mono_resolution][1]);

  right_publisher_ =
    std::make_unique<dai::rosBridge::BridgePublisher<sensor_msgs::msg::Image, dai::ImgFrame>>(
      right_queue, node_, plugin_name_ + std::string("/right/image"),
      std::bind(
        &dai::rosBridge::ImageConverter::toRosMsg, right_converter_.get(), std::placeholders::_1,
        std::placeholders::_2),
      kQueueSize, right_camera_info, plugin_name_ + "/right");

  if (camera_params_.enable_infra) {
    left_publisher_->addPublisherCallback();
    right_publisher_->addPublisherCallback();
  }

  // Add rgb publisher
  if (camera_params_.enable_rgb) {
    auto rgb_queue = device_->getOutputQueue("rgb", 30, false);
    rgb_converter_ =
      std::make_unique<dai::rosBridge::ImageConverter>(plugin_name_ + "_rgb_frame", true);

    auto rgb_camera_info = rgb_converter_->calibrationToCameraInfo(
      calibration_handler, dai::CameraBoardSocket::RGB,
[... 216 lines cut ...]
[robot_state_publisher-1] [INFO] [1665386921.293684124] [oak_state_publisher]: got segment front_camera_right_camera_optical_frame
[component_container-2] [INFO] [1665386921.502091933] [depthai_container]: Load Library: /home/akshay/cleaning_peppermint_ws/install/depthai_ros_ppmt/lib/libmulti_cam_interface_comp_node.so
[component_container-2] [INFO] [1665386921.503405640] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<ppmt_multi_cam_driver::MultiCamInterfaceCompNode>
[component_container-2] [INFO] [1665386921.503455780] [depthai_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ppmt_multi_cam_driver::MultiCamInterfaceCompNode>
[component_container-2] [INFO] [1665386921.513378916] [multi_cam_interface_composable]: Multi Cam Interface Comp Node Initialised
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/multi_cam_interface_composable' in container '/depthai_container'
[component_container-2] [INFO] [1665386921.514811681] [depthai_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[component_container-2] [INFO] [1665386921.554824832] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1665386921.554872440] [depthai_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/front_camera_convert_metric_node' in container '/depthai_container'
[component_container-2] [INFO] [1665386921.578150799] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1665386921.578178966] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1665386921.578183887] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1665386921.578188094] [depthai_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1665386921.578192010] [depthai_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/front_camera_point_cloud_xyz_node' in container '/depthai_container'
[component_container-2] Multiple camera interface plugin loaded!
[component_container-2] [INFO] [1665386922.523556377] [multi_cam_interface_composable]: Initialising Multiple Camera Interface Plugin for 'front_camera'
[component_container-2] [INFO] [1665386922.527422413] [multi_cam_interface_composable]: Camera device with MXID 19443010E1D4F91200 found!
[component_container-2] [INFO] [1665386925.838804488] [multi_cam_interface_composable]: OAK-D device for camera 'front_camera' with MXID : 19443010E1D4F91200, is configured!
**[component_container-2] /opt/ros/humble/lib/rclcpp_components/component_container: symbol lookup error: /home/akshay/cleaning_peppermint_ws/install/depthai_bridge/lib/libdepthai_bridge.so: undefined symbol: _ZN3dai18CalibrationHandler19getCameraIntrinsicsENS_17CameraBoardSocketEiiNS_7Point2fES2_**
[ERROR] [component_container-2]: process has died [pid 60123, exit code 127, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=depthai_container -r __ns:=/'].

 - Which OS/OS version are you using? - Ubuntu 22.04
 - Which ROS version are you using? - ROS2

saching13, 2022-10-17:
Please append the RME. aka the package in ROS case.  
https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example

Serafadam, 2022-11-09:
Basing on this description, I would check if all necessary headers are included, verify CMakeLists.txt and package.xml in your example

closed 2023-09-11 by Serafadam