luxonis/depthai-ros #158: [Feature-Request] {Tf publishers}

an99990, 2022-10-30:
### Start with the `why`:
I am recording a ros bag of the different topics i need from the camera to run rtabmap. I am encoutering some issues because theres some tfs missing. When i record i end up with only one message for the tf_static. 
Is there a way to get tfs

[image]

saching13, 2022-11-01:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_slam/launch/turtlebot3_manipulation_slam.launch
Even though they mention the freq here. it would still be published only once since it is static. it will publush repetatively if you change from static. I reviewed how the robot_state_publisher is written.

an99990, 2022-11-01:
What do you mean if i change from static ?
Currently I'm having issues with RTABMAP because it can't find the transform at time t.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_slam/launch/turtlebot3_manipulation_slam.launch

an99990, 2022-11-01:
> @an99990<https://github.com/an99990> TF is published using robot_state_publisher here<https://github.com/luxonis/depthai-ros/blob/af3a1ebff8d3ec8eecdece344d1250805294d520/depthai_bridge/launch/urdf.launch#L32> and since it is a static transform you don't need to repetatively subscribe. But if doing VIO you will ahve a map/odom frame w.r.t which this total frame moves. And making this keep publishing again and again the same static transform is waste of bandwidh. So I think hence the robot_state_publisher is designed this way ?
> please correct me if I'm wrong here.

Is the robot state publisher launched by default when I launch the stereo_inertial_publisher for example ?

saching13, 2022-11-01:
> What do you mean if i change from static ?
> Currently I'm having issues with RTABMAP because it can't find the transform at time t.
> 
> From: Sachin Guruswamy ***@***.***>
> Sent: Tuesday, November 1, 2022 7:33:57 AM
> To: luxonis/depthai-ros ***@***.***>
> Cc: an99990 ***@***.***>; Mention ***@***.***>
> 
> 
> https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_slam/launch/turtlebot3_manipulation_slam.launch
> Even though they mention the freq here. it would still be published only once since it is static. it will publush repetatively if you change from static. I reviewed how the robot_state_publisher is written.

It means it will be published in /tf.
I will add an option to change. 
If I understand what rtabmap is trying to do is get and change the replative from map to base of oak.

saching13, 2022-11-01:
> Is the robot state publisher launched by default when I launch the stereo_inertial_publisher for example ?
> From: Sachin Guruswamy ***@***.***>
> Sent: Tuesday, November 1, 2022 7:08:24 AM
> To: luxonis/depthai-ros ***@***.***>
> Cc: an99990 ***@***.***>; Mention ***@***.***>
> 
> 
> @an99990<https://github.com/an99990> TF is published using robot_state_publisher here<https://github.com/luxonis/depthai-ros/blob/af3a1ebff8d3ec8eecdece344d1250805294d520/depthai_bridge/launch/urdf.launch#L32> and since it is a static transform you don't need to repetatively subscribe. But if doing VIO you will ahve a map/odom frame w.r.t which this total frame moves. And making this keep publishing again and again the same static transform is waste of bandwidh. So I think hence the robot_state_publisher is designed this way ?
> please correct me if I'm wrong here.

Yes.

saching13, 2022-11-01:
Have you checked the issue in rtabmap ?

saching13, 2022-11-01:
https://github.com/luxonis/depthai-ros/issues/53#issuecomment-1297052725
Check this for rtabmap issue.

an99990, 2022-11-01:
You wrote this :  it will publush repetatively if you change from static.
And I'm asking what it means.

Thank you

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_slam/launch/turtlebot3_manipulation_slam.launch

an99990, 2022-11-02:
but i cant record tf topics in a rosbag, how would i get the transforms ?

saching13, 2022-11-03:
> You wrote this : it will publush repetatively if you change from static. And I'm asking what it means. Thank you
> […](#)
> ________________________________ From: Sachin Guruswamy ***@***.***> Sent: [...]

check [here](http://wiki.ros.org/robot_state_publisher). If you set the `use_tf_static=False` then it will publish on `/tf` at 50Hz.

saching13, 2022-11-03:
> but i cant record tf topics in a rosbag, how would i get the transforms ?

it is on `/tf_static` but it is latched. Since it is a static robot model I'm not sure why you need it to keep updating frequently.

an99990, 2022-11-03:
the issue was fixed with the latest merge in depth-core ai , thanks

an99990, 2022-11-03:
actually i just tried it again and it still doesnt work: opened the following issue : https://github.com/introlab/rtabmap_ros/issues/835https://github.com/introlab/rtabmap_ros/issues/835

peci1, 2022-11-09:
Maybe you're on ROS melodic and you have more static TF publishers? If that's the case and you're replaying a bag file, it doesn't work: https://github.com/ros/ros_comm/issues/146 (this issue has been fixed in Noetic). If you're using Melodic, just install `ros-melodic-static-transform-mux` , open a console and run `rosrun static_transform_mux static_transform_mux` in it. After that, start replaying the bag file. Now you should have all TFs.

saching13, 2023-07-22:
Closing this issue since it looks like resolved. Feel free to open it if not.

closed 2023-07-22 by saching13