luxonis/depthai-ros #159: Ros Coordinate System And OAK Coordinate System Not Matching/

Mani-Radhakrishanan, 2022-11-03:
https://www.ros.org/reps/rep-0103.html 

The above Standards Followed??When we integrate the system with ros (Moveit) the target(Aruco) is in random position.

Mani-Radhakrishanan, 2022-11-03:
We are using Moveit for robot arm calibration.The target frame is Aruco Marker.We are following standard ROS coordinate system in the our URDF. But the image coming out of depthai is not aligning with ROS Coordinate System(RIGHT HAND COORDINATE SYSTEM) .We are not facing the issue with realsense..

Is there anywhere in the depthai code mentionded about coordinate system ??

saching13, 2022-11-04:
> We are using Moveit for robot arm calibration.The target frame is Aruco Marker.We are following standard ROS coordinate system in the our URDF. But the image coming out of depthai is not aligning [...]

If you are referring to the depth frames. It is in mm from the DepthAI. We need to change that to meters for the ROS. And focal length is in pixel units.

Mani-Radhakrishanan, 2022-11-04:
[image]
  

Can You explain What is the Coordinate system mentioned in the Image???  Is there a Convention followed on coordinate frame???

Serafadam, 2022-11-07:
Hi, you can view all the frames in Rviz by enabling TF option. You can find that for a camera there is a regular frame and an `optical` frame, here is an example:
[image]
optical frame represents image in the [opencv format](https://stackoverflow.com/questions/17987465/how-is-the-camera-coordinate-system-in-opencv-oriented). This is a similar convention to the one that Gazebo uses. We'll be probably adding an option to switch between the two, but for now you can either republish the image in different frame or modify the code to switch to the frame needed.

saching13, 2022-11-09:
> dai_rgb_camera_optical_frame to camera_link (in my robot)
You should do it to the base of the camera.

> Also Can you explain the coorindate frame mentioned in the PCB???

This is IMU mounting orientation. If you look at it. it does follow the Right-hand rule.

Mani-Radhakrishanan, 2022-11-13:
No it is working for me. So which point is the frame of reference in urdf ,RGB or One of the Stereo Camera.
(i.e) Which camera point is reference for XYZ Coordinates ???Currently I am taking RGB as my reference point but it seems to be there is 37.5 shift in X ???Is it anything to do with Right Stereo Camera?

Serafadam, 2022-11-14:
Base mounting point should be `oak-d-frame`. Could you show us your TF tree? Or a screenshot from rviz, maybe it could better illustrate what we are dealing with.

Mani-Radhakrishanan, 2022-11-15:
[image]
why the target frame is in weird pose? I am using rgb_stero_node.launch .I did the static transform between the dai_rgb_camera_optical_frame to camera_link in my robot....Getting this weird pose.

I just came to know that stereo_inertial_node.launch is the proper one??Is it right ?

Serafadam, 2022-11-16:
Hi, yes, `stereo_inertial_node` is the one that is most up to date, can you post the results when running it?

Mani-Radhakrishanan, 2022-11-17:
I tried with stereo_inertial_node still I am getting the same result as I mentioned in above query.Why the  target position is different location ??? I am putting the target in front but it shows in the side ?? Any sugestions are appreciated
[image]

asimonov, 2023-09-19:
> [image]
> 
> Can You explain What is the Coordinate system mentioned in the Image??? Is there a Convention followed on coordinate frame???

thanks for the picture!
yes, it looks like this is back of the PCB on which IMU is mounted in my OAK-D Pro W.

so IMU coordinate frame is: x - down, y - rght, z - backwards if looking at the camera from behind