luxonis/depthai-ros #160: [BUG] weird pose

an99990, 2022-11-03:
Hi,
I am using the depth information to calculate the pose of detected objects in the scene, i have few questions.
What is the camera frame ? is there an image showing whats the x,y,z axis ? ALso when i calculate the centroids, i get the same value for x,y,z coordinates, i have teste the same code with another camera but didnt get this issue.

I leave the code here, please let me know if the issue is on my end.
thank you

```
center_x = camera_model.cx() * scale
center_y = camera_model.cy() * scale
 unit_scaling = DepthTraits.to_meters(1)
constant_x = unit_scaling / (camera_model.fx() * scale)
constant_y = unit_scaling / (camera_model.fy() * scale)
        for mask in depth_mask:
            if mask is None:
                continue
            centroid_pose = np.zeros(3).astype(np.float32)
            n_valid_points = 0
            for j in range(depth_image.shape[0]):
                for i in range(depth_image.shape[1]):
                    if mask[j][i] == 0:
                        continue
                    depth = depth_image[j][i]
                    if not DepthTraits.valid(np.array(depth)):
                        continue
                    depth = min(depth, range_max) if range_max > 0 else depth

                    depth_point = np.array([
                        (i - center_x) * depth * constant_x,
                        (j - center_y) * depth * constant_y,
                        DepthTraits.to_meters(depth)])
                    centroid_pose += depth_point[0]
                    n_valid_points += 1

            if n_valid_points == 0:
                centroid_pose = np.array([np.nan, np.nan, np.nan])
            else:
                centroid_pose /= n_valid_points
            centroids.append(centroid_pose)
        return centroids
```

Mani-Radhakrishanan, 2022-11-07:
I did the static transform between the two coordinate frame using tf frames in ROS

dai_rgb_camera_optical_frame to camera_link (in my robot)

Also edited the code to match my link name in urdf. But still the detected target(Aruco) is in weird place.

https://homepages.inf.ed.ac.uk/rbf/CVonline/LOCAL_COPIES/OWENS/LECT9/node2.html
We have followed the same convention in our code???

Is anywhere coordinate frame convention is defined in depthai???

Also Can you explain the coorindate frame mentioned in the PCB???

an99990, 2023-02-28:
thank you for posting the issue here @Serafadam but i am not sure how this is link with mine, I have weird resultst with the depth image from the stereo topic. I have tried my code with the zed camera and it works fine. the pose is also at the top corner on the left side for some reason.