luxonis/depthai-ros #170: Feature request: Enable/disable camera through ROS2 service

saching13, 2022-11-19:
It would be useful to be able to enable/disable the camera streams at runtime or put the camera into a low power state when not in use, without actually killing the ROS2 node. This could be done with a [SetBool](http://docs.ros.org/en/api/std_srvs/html/srv/SetBool.html) service.
Moving from [here](https://github.com/luxonis/depthai-ros-examples/issues/59)

saching13, 2022-11-19:
Instead of setbool. How about using a lifecycle node ?

Serafadam, 2022-11-21:
I'd say lifecycle nodes are on the list, but until [this clears out](https://github.com/ros-perception/image_common/issues/108) we can add a service

closed 2023-07-04 by Serafadam

silviazpi, 2023-07-04:
Why did you close it? Is it already added?

Serafadam, 2023-07-04:
Hi, yes, camera start/shutdown services have been available for some time now