luxonis/depthai-ros #171: Different fps for video and preview output

saching13, 2022-11-19:
Is it possible to modify the code such that the frame rates for video and preview output are different? I'm using the camera on a resource contraint platform and want to lower the computational burden. As an example I would like to have 30 fps for the preview output and 2 fps for the video output.

Alternatively, could there be a ROS service to capture still images at the full sensor resolution?

Moving from [here](https://github.com/luxonis/depthai-ros-examples/issues/61)

saching13, 2022-11-19:
cc: @alexswerner

saching13, 2022-11-19:
@alexswerner Yes. that should be possible. 
Either you can create a script node and pass the video output to throttle and send at 2fps. 
Or you can use throttle tools like [this](http://wiki.ros.org/topic_tools/throttle) 
Or like you mentioned you can also have a action or service to make it capture at full res.

alexswerner, 2022-11-28:
Just got back to this. Thanks for the reply. When you write _script node_, you mean a python script which republishes the messages at a lower rate? I'm not sure if that would serve the purpose of reducing the computational (i/o) burden on the cpu as the camera whould then still transfer the full-size images at full rate. Or am I mistaken?