luxonis/depthai-ros #173: [BUG] Enable to access Point cloud/image in ROS noetic with OAK-D Pro Poe camera (Ubuntu 20.04)

harshkakashaniya, 2022-11-22:
We are able to access depthai_demo.py and also most of the scripts in depthai-python repo.

Can we get assistance with the following issue?

When we run : 
```roslaunch depthai_examples stereo_inertial_node.launch```

This is the output from the terminal.

```... logging to /home/hbk/.ros/log/41e4c682-6a82-11ed-b084-3d427a22a2e4/roslaunch-harsh-il-105488.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://harsh-il:44519/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/detectionClassesCount: 80
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/expTime: 20000
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/manualExposure: False
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 19443010F10C731300
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: True
 * /stereo_inertial_publisher/previewHeight: 416
 * /stereo_inertial_publisher/previewWidth: 416
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/hbk/oakd_pr...
 * /stereo_inertial_publisher/rgbResolution: 1080p
 * /stereo_inertial_publisher/rgbScaleDinominator: 3
 * /stereo_inertial_publisher/rgbScaleNumerator: 2
 * /stereo_inertial_publisher/sensIso: 800
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

auto-starting new master
process[master]: started with pid [105521]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 41e4c682-6a82-11ed-b084-3d427a22a2e4
process[rosout-1]: started with pid [105542]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [105549]
process[stereo_inertial_publisher-3]: started with pid [105550]
process[nodelet_manager-4]: started with pid [105551]
process[depth_image_convertion_nodelet-5]: started with pid [105552]
process[depth_image_to_rgb_pointcloud-6]: started with pid [105558]
[ INFO] [1669134321.228269819]: Initializing nodelet with 12 worker threads.
 nnPath ,, /home/hbk/oakd_pro/depthai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
1280 720 1280 720
Listing available devices...
Device Mx ID: 19443010F10C731300
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 105550, exit code -6, cmd /home/hbk/oakd_pro/depthai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/hbk/.ros/log/41e4c682-6a82-11ed-b084-3d427a22a2e4/stereo_inertial_publisher-3.log].
log file: /home/hbk/.ros/log/41e4c682-6a82-11ed-b084-3d427a22a2e4/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

We did some digging in the code of stereo_intetial_publisher.cpp and code is getting stuck at line 389

```
device = std::make_shared<dai::Device>(pipeline, deviceInfo);
```

Thank you in advance for the solution.

saching13, 2022-11-23:
Can you check if it works by unplugging and plugging the device back after running depthai_demo.py ?

harshkakashaniya, 2022-11-23:
I was able to make it work by changing the IP address.
Not sure what was the issue.

Thank you very much @saching13 

Now when I am running this package in a container I am not able to access the point cloud it says no cameras are available.
I am using the bridge network in docker.

Can we specify an IP address instead of Mx ID  to search for the camera?

saching13, 2022-11-28:
@harshkakashaniya  yes. you can use the IP to search for the device. 
there is a on-going PR for that [here](https://github.com/luxonis/depthai-ros/pull/153)