luxonis/depthai-ros #177: errors in roslaunch of depthai_examples

Kazimbalti, 2022-12-02:
I am having Ubuntu 20.04 with Noetic and python 3.
catkin_make also build successfully but  
When I launch, it gets the following errors:

roslaunch depthai_examples stereo_inertial_node.launch

... logging to /home/asl/.ros/log/e88e08fa-71ed-11ed-9a87-c742c8a52e19/roslaunch-asl-ThinkPad-P15v-Gen-2i-28065.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

substitution args not supported:  No module named 'rospkg'
when processing file: /home/asl/dai_ws/src/depthai-ros/depthai_bridge/urdf/depthai_descr.urdf.xacro
RLException: while processing /home/asl/dai_ws/src/depthai-ros/depthai_bridge/launch/urdf.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '/home/asl/dai_ws/src/depthai-ros/depthai_bridge/urdf/depthai_descr.urdf.xacro', 'camera_name:=oak', 'camera_model:=OAK-D', 'base_frame:=oak-d_frame', 'parent_frame:=oak-d-base-frame', 'cam_pos_x:=0.0', 'cam_pos_y:=0.0', 'cam_pos_z:=0.0', 'cam_roll:=0.0', 'cam_pitch:=0.0', 'cam_yaw:=0.0']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro '$(find depthai_bridge)/urdf/depthai_descr.urdf.xacro'                         camera_name:=$(arg tf_prefix)                         camera_model:=$(arg camera_model)                         base_frame:=$(arg base_frame)                         parent_frame:=$(arg parent_frame)                         cam_pos_x:=$(arg cam_pos_x)                         cam_pos_y:=$(arg cam_pos_y)                         cam_pos_z:=$(arg cam_pos_z)                         cam_roll:=$(arg cam_roll)                         cam_pitch:=$(arg cam_pitch)                         cam_yaw:=$(arg cam_yaw)"/>
The traceback for the exception was written to the log file

Serafadam, 2022-12-05:
Hi, this seems related more to ROS than to `stereo_intertial_node` itself. You can try methods mentioned here:
 https://answers.ros.org/question/245967/importerror-no-module-named-rospkg-python3-solved/
 if not, if you check out `noetic` branch there is a Dockerfile which allows you to build & run `stereo_inertial_node` isolated.

Serafadam, 2023-01-31:
Closing due to inactivity

closed 2023-01-31 by Serafadam