luxonis/depthai-ros #178: roslaunch error on stereo_initial_node.launch

wanjunkim, 2022-12-02:
my platform is: Xavier AGX + OAKD_Pro
When I launch this
roslaunch depthai_examples stereo_inertial_node.launch

I have get the following error

rocess[rviz-6]: started with pid [44989]
[ INFO] [1669977447.974530422]: Initializing nodelet with 8 worker threads.
 nnPath ,, /home/f1tenth_asl/catkin_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
1280 720 1280 720
Listing available devices...
Device Mx ID: 18443010F184641200

[ERROR] [1669977449.458188006]: PluginlibFactory: The plugin for class 'rviz_imu_plugin/Imu' failed to load.  Error: According to the loaded plugin descriptions the class rviz_imu_plugin/Imu with base class type rviz::Display does not exist. Declared types are  rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_plugin_tutorials/Imu

[18443010F184641200] [1.1.1] [1.648] [SpatialDetectionNetwork(9)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: SUPER
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 44975, exit code -11, cmd /home/f1tenth_asl/catkin_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/f1tenth_asl/.ros/log/95972912-722c-11ed-8ca7-b0359fb57a75/stereo_inertial_publisher-2.log].
log file: /home/f1tenth_asl/.ros/log/95972912-722c-11ed-8ca7-b0359fb57a75/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Serafadam, 2022-12-05:
Hi, it seems you need to install additional packages - in this case
`sudo apt install ros-noetic-rviz-imu-plugin` should solve the issue

Serafadam, 2023-01-31:
Closing due to inactivity

closed 2023-01-31 by Serafadam