luxonis/depthai-ros #180: Pi4 4g stalls when running ros(noetic) node

Tjaart-Hobby, 2022-12-14:
**Check if issue already exists**
could not find similar issues.

Running the node on laptop (ubuntu 20.04, noetic) installed with "apt install ros-noetic-......." works but when running node on Pi4 4g and 2g swap file, with ubuntu 20.04/server + xcfe desktop ,purely to test gui's,camra things, the node starts great but 1/2 min later Pi4 becomes unresponsive to the point where I have to remove power to it. It wont even shutdown.
Note: plan to run node on pi4 and ros network to laptop that will do the work.

Fixing this might fix Imu not working in rviz(working on this issue now) I just "sudo apt install ros-noetic-rviz-imu-plugin" on laptop not tested yet.

**Expected behavior**
Pi4 to run node /publishes topics/ and ros-networked laptop will do rviz and rqt-image-view to drive robot with.
note: after installing node on laptop I had issues with rqt-image-view. I had to "pip3" uninstall pyqt5 and pyqt5-sip to get rqt/robot steer/image view and all other rqt related things to work again. pyqt5 were already "apt" installed on machine.
Note camera working just fine with Pi it only crashes when node is run.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - can't do above > pi crashes, will have a look next time I start up the pi, where do I find this "log"

**Additional context**
I just bought this expensive (to me) hardware on a hobbyist budget and would greatly appreciate any help.
Would compiling from source healed any better results?

Serafadam, 2022-12-15:
Enabling desktop environment for RPi takes a lot of resources from it in base configuration, processing/displaying streams straight on RPi also doesn't help. 
Last time I had issues with performance on RPI, what helped was [overclocking](https://beebom.com/how-overclock-raspberry-pi-4/), though with that you need some case with fans/heatsinks to protect it from overheating (like the one [here](https://www.elektor.com/joy-it-armor-aluminium-heatsink-case-with-dual-fan-for-raspberry-pi-4), you can find it in most hobby shops). 
Also, increasing swap will also help.
As for the `rviz-imu-plugins`, it's not yet fixed on apt.

Tjaart-Hobby, 2022-12-15:
I had/have  the same issue using the Pi camera. Pi camera works fine on 32bit OS but with the 64bit ,it starts up fine and crashes the Pi. I couldn't fix it but from memory, had something to do with MMAL and 64bit OS. I got it to working with using a different driver v4gl(something like that) or openCV.  

I changed to Xbox kinect v1 cameras and got them working. They work great but mapping and object recognition is painfully slow( 9fs). Work load is handled by pi4. Still the pi doesn't crash.

If I were to guess. I think your ros node publishes continuously and fills up memory of pi. Not emptying the buffer as new frames come in?
rviz-imi-plugins > work on laptop , not tested on Pi(been trying to solve this crashing)

Serafadam, 2022-12-16:
I ran the node on my laptop for some time but didn't see any substantial memory increase. I'll try it on RPi in the next week, but I suspect that the problem lies somewhere else.

brianlmerritt, 2023-01-02:
A potential cause is power brownout - I've just ordered a powered USB hub (MEGA4 - 4-Port USB 3.1) so I can have an external power for the PI and all peripherals with no danger of burning out the PI and / or running low on power.

Tjaart-Hobby, 2023-01-02:
Hi, unfortunately I am away for the holidays and will only be home after
5/1/2023

I tried the other exsamples and they mostly work fine. It seems the
stereo_inertial_node.launch.py are only one that locks up the Pi.

I am using a oak-d series 2 if that helps

I also created a second workspace and tried to build from source, but that
errored out.

when I am back I will upload a video of the issue