luxonis/depthai-ros #199: Unable to build in ROS-Melodic

GerardHarkemaAvans, 2023-01-13:
Dear reader,

at the moment it is not possible to build the distribution for the ROS1-melodic distribution. See posts below when building.

[build] Warning: Skipping package `depthai-ros` because it has an unsupported package build type: `ament_cmake`
[build] Note: Available build types:                                           
[build]  - `catkin`                                                            
[build]  - `cmake`

Serafadam, 2023-01-13:
Hi, for ROS1 we support only Noetic, if you have to use Melodic, then I'd suggest running the driver inside Docker container.

GerardHarkemaAvans, 2023-01-13:
Too bad,

in the mid-december clone it did work with Melodic. We integrate the camera with NirioNED educational robots, which only supports ROS Melodic.

Thank you for the reply.

saching13, 2023-01-13:
I think you tried on humbe branch.
Can you try on noetic or ros-release branch ?

Serafadam, 2023-01-31:
Hi @GerardHarkemaAvans, you can try using docker images for Noetic available [here](https://hub.docker.com/r/luxonis/depthai-ros) and see if that works with your system

MoscowskyAnton, 2023-03-03:
Faced same error, checking out to ros-release branch helps build it on Melodic

GerardHarkemaAvans, 2023-03-05:
I made a working copy for our own students on melodice, you can clone it here: https://github.com/GerardHarkemaAvans/depthai_ros.git

manikjslb, 2023-06-22:
> I made a working copy for our own students on melodice, you can clone it here: https://github.com/GerardHarkemaAvans/depthai_ros.git

how did you manage to get the SDK install?
can you share the steps to build it with melodic?

I a, getting following error when I try to use your branch with melodic

[image]