luxonis/depthai-ros #201: Link a robot to OAK-D

metanav, 2023-01-16:
I am trying to connect a robot's base_link to the OAK-D oak_rgb_camera_optical_frame. I can calculate the base_link distance from the camera in the 3D manually. What is the following configuration for? Is it relative to the base_link or I can use the orientation from the IMU (after converting quaternion to Euler)?

depthai_examples/launch/yolov4_publisher.launch.py:

```
    cam_roll  = LaunchConfiguration('cam_roll',      default = '0.0')
    cam_pitch = LaunchConfiguration('cam_pitch',     default = '0.0')
    cam_yaw   = LaunchConfiguration('cam_yaw',       default = '0.0')
```

Serafadam, 2023-01-16:
Hi, 
You can check out current URDF representation [here](https://github.com/luxonis/depthai-ros/blob/humble/depthai_bridge/urdf/include/depthai_macro.urdf.xacro)
For connecting cameras to the robot you'll want to use `parent_frame` (which defaults to `oak-d-base-frame`). As for the configuration, currently these represent RPY orientation of `base frame`  to the `parent frame`. See how [multicam](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/example_multicam.launch.py)`example for `depthai_ros_driver` uses those parameters.
Also, are you using mobile robot or manipulator? In case of manipulator you might also want to check out [hand-eye calibration](https://ros-planning.github.io/moveit_tutorials/doc/hand_eye_calibration/hand_eye_calibration_tutorial.html)

metanav, 2023-01-16:
Thanks, @Serafadam. I will check the new URDF. By the way, I am using a manipulator with the camera mounted overhead at 55 degree (pitch). 
[image]

metanav, 2023-01-17:
I was able to connect the manipulator base_link to the OAK-D using the following command line.

`ros2 launch depthai_examples  yolov4_publisher.launch.py cam_pos_x:=0.90 cam_pos_y:=0 cam_pos_z:=0.70 cam_roll:=0.0 cam_pitch:=0.610865  cam_yaw:=3.14159 parent_frame:=base_link`

Now the problem is the transform seems not working from the `oak_rgb_camera_optical_frame` to `base_link`.

Serafadam, 2023-01-17:
Can you display whole TF tree in Rviz?

metanav, 2023-01-17:
Below is the whole TF tree. in the Rviz. I have removed the name since it was not readable.
[image]

metanav, 2023-01-18:
@Serafadam Can you check what is the pose of the detected object with respect to base_link frame for the above setup which you can create easily in your environment? I think the TF conversion issue is with the tilted camera only. I have checked many other projects but they have the camera in straight position. BTW what is the factory camera calibration pose? Do I need to recalibrate it if I use the camera tilted?

metanav, 2023-01-18:
There is some improvement after changing the tilt angle to 20 degrees, introducing some background with textures, and elevating the objects a bit from the background (floor in my case). In the base_link frame, the y position seems close and the x position with a 15-20 mm error with different distances. The z position is not accurate.

metanav, 2023-01-23:
@Serafadam, I have used depthai-python/examples/SpatialDetection/spatial_location_calculator.py to compare.
x: -0.154m, y: -0.072m, z: 0.642m

ROS:
x: -0.122m, y: -0.085m, z: 0.643m

I think the difference in 'x' is due to mono right vs rgb center reference frame, otherwise they are almost similar.
If I set rotation arguments to zero, the X axis in the camera frame and Z axis in the base link frame have correct transformation values but the Z axis in the base link frame is totally wrong.

metanav, 2023-01-23:
I did some testing with roll = 0, pitch = 0 and yaw= 3.14159. So there is no tilt in the testing except the camera is rotated 180 degrees to face the base_link as shown in the Rviz in the previous posts. I strongly feel that `oak_rgb_camera_optical_frame's` `Y` coordinate value does not correspond to the axis direction and seems shifted as well. I guess spatial calculation is just being assigned and not considering the axis. This could be the root cause for the wrong TF transform for Y that corresponds to the Z in the base_link frame.

metanav, 2023-01-24:
Maybe the tilt angle is not correct as I am measuring. I tried to set the angle  with the results from the IMU rotation vector depthai-python example but the quaternions seems not accurate.  Is there a way to calibrate IMU?

jide07, 2024-07-29:
Hello @metanav  you mind if i asked how you were able to connect your transform tree of panda and oak-d-lite camera, I'm trying to transform the camera frame to base frame for a pick and place operation

metanav, 2024-07-29:
Using the params;  parent_frame:=base_link.

jide07, 2024-07-29:
Where will i do that, this is how my tree looks like
[image]

metanav, 2024-07-29:
Maybe try with parent_frame:=panda_link7. You should pass the params to your launch file.

For example: 
```
$ ros2 launch depthai_examples  yolov4_publisher.launch.py cam_pos_x:=0.90 cam_pos_y:=0 cam_pos_z:=0.70  \
cam_roll:=0.0 cam_pitch:=0.610865  cam_yaw:=3.14159 parent_frame:=panda_link7
```

jide07, 2024-07-29:
I can see something like this now, which means it is connected
[image]

I have placed my camera on the end-effector how can I calculate the camera parameters in relation to that?