luxonis/depthai-ros #205: Setting AutoFocus off in ROS Foxy

divyanshurs, 2023-01-17:
Hi, we are using the ROS-Foxy branch with the depth-ai camera and are looking for a way to turn off the AutoFocus of the camera through launch files (or other setting in the driver). Could you please let us know how we can do this? 

Thanks

fjahncke, 2023-01-17:
To specify we are looking for a way to set this in the rgb_publisher.launch.py or rgb_publisher.cpp

Serafadam, 2023-01-18:
Hi, currently there are two options as `depthai_ros_driver` is not (yet) available on ROS Foxy:

1. Edit source code or  create your own publisher using BridgePublishers & Image converters - `stereo_inertial_node` is a good example of this. To set manual focus, just call `colorCam->initialControl.setManualFocus(focus);`
2. Build repo Humble version from source or in Docker, there should also be docker images pushed to Docker Hub this week  to download & play with.
Foxy & Galactic versions of repository will be updated this/next week depending on how fast that goes.

fjahncke, 2023-01-18:
Thank you that solved the problem. For anyone looking for the precise changes, just in rgb_publisher.cpp after line 23 I added 
`colorCam->initialControl.setManualFocus(130);`

It is already the Foxy branch and works perfectly.

Serafadam, 2023-01-31:
Seems ok to close, also there are new images for Humble that should be ok to use in Foxy [here](https://hub.docker.com/r/luxonis/depthai-ros)

closed 2023-01-31 by Serafadam