luxonis/depthai-ros #210: ros noetic stereo_inertial_node and depthai_ros_driver/pointcloud.launch error

youngjae41, 2023-01-20:
Jetson orin AGX + Oak-D-PRO 

depthai_ros_driver/pointcloud.launch error

[image]

stereo_inertial_node launch error

[image]

Serafadam, 2023-01-20:
Hi, `pointcloud.launch` should be fixed by https://github.com/luxonis/depthai-ros/pull/211, as for the `stereo_intertial_node`, can you try adding `export OPENBLAS_CORETYPE=ARMV8` to your `bashrc` file?

youngjae41, 2023-01-25:
@Serafadam Thanks !! works well

youngjae41, 2023-01-25:
@Serafadam  Hello Serafadam

desktop works well 
but jetson not work!

youngjae41, 2023-01-25:
@Serafadam

I'm found problem

it was cv_brige issue

changed

/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake 
change this line:

set(_include_dirs "include;;/usr/include/opencv4") 
to

set(_include_dirs "include;/usr/local/include/opencv4")

closed 2023-01-25 by youngjae41