luxonis/depthai-ros #214: [Calibration] Recalibrate the OAK-D

jonathanleinola, 2023-01-29:
Do I need to recalibrate the OAK-D? It seems that the point cloud is going through the walls. I tried to launch with [depthai_ros_driver] using the Noetic branch. If i need to calibrate it then how could i do it?

jonathanleinola, 2023-01-29:
[image] Here's an image of the issue

Serafadam, 2023-01-30:
Hi, for easier debugging:
1. Can you show output of `stereo_intertial_node`
2. Can you also attach image of the depth stream?

Serafadam, 2023-01-30:
That data look pretty noisy, do you mind sharing also left and right stream data? (this can be enabled by setting parameters in `rgbd.yaml`. You can also try changing `stereo` parameters and see if there are any improvements in your use-case.
`stereo_intertial_node` is in depthai examples, so roslaunch depthai_examples stereo_inertial_node.launch
WLS filtering should be merged this week, and it will help with making PCL a little denser, you can check progress here https://github.com/luxonis/depthai-ros/tree/wls_filter