luxonis/depthai-ros #215: depthai_ros_driver Humble release

doisyg, 2023-01-31:
### Start with the `why`:
For convenience and for ROS users not to have to deal with building it from source

### Move to the `what`:
Releasing built binaries of `depthai_ros_driver` to current ROS2 LTS, i.e. Humble

### Move to the `how`:
Same as for the other packages

Any plans ? Anything blocking ?

ladrians, 2023-01-31:
+1 to the request, for ROS1 Noetic would be great too, using a rpi4 it takes a lot of time on every compile cycle.

saching13, 2023-02-01:
we released it last week.
[here](https://github.com/ros/rosdistro/pull/35978#event-8391358227) is the PR for Humble release.
[here](https://github.com/ros/rosdistro/pull/35979#event-8401517187) for the Noetic release

saching13, 2023-02-03:
Yes. I noticed it yesterday. Sent a PR to enable pr testing on it. 
Will have new version up in 2 days.
Sorry for the issue.

doisyg, 2023-02-07:
@saching13 any updates ? (sorry for pushing but it is quite heavy to build from src, @scheunemann)

saching13, 2023-02-07:
Sorry for the delay. @doisyg 
PR testing is still not working for some reason. I have requested Tully to see what is the issue. 
That said I sent out a release PR either way [here](https://github.com/ros/rosdistro/pull/36058). Hoping to be merged today.

saching13, 2023-02-08:
@doisyg  @roni-kreinin  release is up and looks like it's stable.
https://build.ros2.org/job/Hbin_ujv8_uJv8__depthai_ros_driver__ubuntu_jammy_arm64__binary/

Serafadam, 2023-03-17:
Packages are available for Humble, closing issue.

closed 2023-03-17 by Serafadam