luxonis/depthai-ros #217: Filter camera by USB port id

doisyg, 2023-02-01:
### Start with the `why`:
For multiple cameras systems in production, it becomes quickly complex to manage a long list of camera serial numbers (current filter with `i_mx_id`). An alternative could be to filter by USB port id as it is done classically for cameras in ROS, see for instance `usb_port_id` in https://github.com/IntelRealSense/realsense-ros

### Move to the `what`:
Add a parameter `usb_port_id` to `depthai_ros_driver::Camera` for the driver to open only the camera plugged to the corresponding port

### Move to the `how`:
@mattmacleod 
> it would mean quite a few changes to depthai-core as well to expose USB bus information in the Device class

Serafadam, 2023-02-01:
Thanks for that request, this shouldn't be a big issue to add it :+1:

Serafadam, 2023-02-16:
Added in https://github.com/luxonis/depthai-ros/pull/229, available on `v2.6.3`

closed 2023-02-16 by Serafadam