luxonis/depthai-ros #220: [catkin_make failed] {ROS Noetic catkin_make failing}

Aloha-Churchill, 2023-02-02:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
I am running Ubuntu 20.04 with ROS Noetic and am trying to set up the Luxonis OAK-D wide camera with ROS. I have followed the instructions for Install from source on the Noetic branch and am running into an error when running the "catkin_make" (step 7) command and have attached the output below. I also noticed a slight typo in step 4 (I think it should be cd .. instead of cd ../..).

[image]

[image]

Thank you so much for your help!

Serafadam, 2023-02-03:
Hi,
Thanks for checking, indeed `catkin_make` commands should be updated to `catkin_make_isolated` which should work without problems. Can you try building with catkin tools and check if you have the same results? https://catkin-tools.readthedocs.io/en/latest/installing.html

Aloha-Churchill, 2023-02-03:
Thanks for your quick response. I've installed catkin tools and ran `catkin build` instead of `catkin_make` on a clean install. This step seemed to run without errors, but produced the following warnings.

[image]

I then proceeded to the next step to first source the setup.bash file and then ran `roslaunch depthai_examples stereo_inertial_node.launch` which outputted the following

[image]

I created another simple workspace and used catkin build and then launched a file to check that the migration from `catkin_make` to `catkin build` was okay, so I'm fairly confident this is not the problem. Maybe the reason it is not finding the packages is due to the previous warnings when running `catkin build`, especially since it skipped building the packages due to the unsupported package build `ament_cmake`? Thank you again for your help. I'm looking forward to being able to test the OAK-D on ROS.

Serafadam, 2023-02-03:
Hi, it seems you are using ROS2 branch, checkout `noetic` and it should build fine

Aloha-Churchill, 2023-02-08:
Yes, thank you this worked!

As a recap here are the steps that ended up working for me for Ubuntu 20.04 running ROS Noetic in case future users find it helpful (nearly identical to documentation).

Install dependencies

The following script will install depthai-core and update usb rules and install depthai devices

`sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash`

if you don't have opencv installed then try `sudo apt install libopencv-dev`

if you don't have rosdep installed and not initialized please execute the following steps:

    `sudo apt install python-rosdep(melodic)` or `sudo apt install python3-rosdep`
```
    sudo rosdep init
    rosdep update
```

Setting up procedure

The following setup procedure assumes you have cmake version >= 3.10.2 and OpenCV version >= 4.0.0. We selected dai_ws as the name for a new folder, as it will be our depthai ros workspace.

```
    mkdir -p dai_ws/src
    cd dai_ws/src
    git clone --branch noetic https://github.com/luxonis/depthai-ros.git
    cd ..
    rosdep install --from-paths src --ignore-src -r -y
    source /opt/ros/noetic/setup.bash
```
    `catkin build depthai_examples` --> the entire build kept on crashing my PC (maybe due to system limitations, but building the examples worked)
    `source devel/setup.bash `

closed 2023-02-08 by Aloha-Churchill