luxonis/depthai-ros #221: [BUG] {Title of the bug}

G-ridge, 2023-02-03:
Hello,

I am trying to setup a OAK-D Lite on ROS Foxy on my Ubuntu 20.04.5 LTS machine.

I am getting an error on "Setting up procedure - Step 7" that I can't seem to find an answer to online.

```
~/Depthai-Tests/ros2-test/dai_ws$ ./src/depthai-ros/build.sh
Build type: RelWithDebInfo, Install_type: symlink-install
Sequential build
Starting >>> depthai_ros_msgs
Finished <<< depthai_ros_msgs [12.8s]                       
Starting >>> depthai_bridge
--- stderr: depthai_bridge                               
/home/cg22628/Depthai-Tests/ros2-test/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp: In member function ‘void dai::ros::SpatialDetectionConverter::toRosVisionMsg(std::shared_ptr<dai::SpatialImgDetections>, std::deque<vision_msgs::msg::Detection3DArray_<std::allocator<void> > >&)’:
/home/cg22628/Depthai-Tests/ros2-test/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp:103:49: error: ‘__gnu_cxx::__alloc_traits<std::allocator<vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >, vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >::value_type’ {aka ‘struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’} has no member named ‘hypothesis’
  103 |         opDetectionMsg.detections[i].results[0].hypothesis.class_id = std::to_string(inNetData->detections[i].label);
      |                                                 ^~~~~~~~~~
/home/cg22628/Depthai-Tests/ros2-test/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp:104:49: error: ‘__gnu_cxx::__alloc_traits<std::allocator<vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >, vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >::value_type’ {aka ‘struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’} has no member named ‘hypothesis’
  104 |         opDetectionMsg.detections[i].results[0].hypothesis.score = inNetData->detections[i].confidence;
      |                                                 ^~~~~~~~~~
make[2]: *** [CMakeFiles/depthai_bridge.dir/build.make:115: CMakeFiles/depthai_bridge.dir/src/SpatialDetectionConverter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/depthai_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< depthai_bridge [27.7s, exited with code 2]

Summary: 1 package finished [40.7s]
  1 package failed: depthai_bridge
  1 package had stderr output: depthai_bridge
  3 packages not processed
```

Would anyone happen to know how to fix this issue?

Thank you for you help.

G-ridge

Serafadam, 2023-02-03:
Hi, are you using `foxy` branch?

Serafadam, 2023-02-06:
Since you're using `build.sh` script I suspect you are on a `humble` branch (it hasn't been ported to `foxy` yet.

aenoca, 2023-03-06:
You need to switch branches locally as well. Indeed, when doing the step `git clone https://github.com/luxonis/depthai-ros.git`, this puts you into the master branch (humble). Go into the directory where you cloned it (depthai-ros/) and write `git switch foxy`. This will switch you to the foxy branch. Then do the other steps.