luxonis/depthai-ros #225: ROS2 FOXY Colcon Build Error

AndresCao, 2023-02-09:
I ran through the tutorial you guys have listed and get errors. tried it on 2 different computers both running Ubuntu 20.04 the same result. I delete the depthai_bridge package and it builds fine.

USER@Computer
[image]
:~/dai_ws$ colcon build
Starting >>> depthai_ros_msgs
Finished <<< depthai_ros_msgs [0.30s]                    
Starting >>> depthai_bridge
--- stderr: depthai_bridge                             
/home/andres/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp: In member function ‘void dai::ros::SpatialDetectionConverter::toRosVisionMsg(std::shared_ptr<dai::SpatialImgDetections>, std::deque<vision_msgs::msg::Detection3DArray_<std::allocator<void> > >&)’:
/home/andres/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp:105:49: error: ‘__gnu_cxx::__alloc_traits<std::allocator<vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >, vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >::value_type’ {aka ‘struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’} has no member named ‘hypothesis’
  105 |         opDetectionMsg.detections[i].results[0].hypothesis.class_id = std::to_string(inNetData->detections[i].label);
      |                                                 ^~~~~~~~~~
/home/andres/dai_ws/src/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp:106:49: error: ‘__gnu_cxx::__alloc_traits<std::allocator<vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >, vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> > >::value_type’ {aka ‘struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’} has no member named ‘hypothesis’
  106 |         opDetectionMsg.detections[i].results[0].hypothesis.score = inNetData->detections[i].confidence;
      |                                                 ^~~~~~~~~~
make[2]: *** [CMakeFiles/depthai_bridge.dir/build.make:115: CMakeFiles/depthai_bridge.dir/src/SpatialDetectionConverter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/depthai_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< depthai_bridge [4.14s, exited with code 2]

Summary: 1 package finished [4.53s]
  1 package failed: depthai_bridge
  1 package had stderr output: depthai_bridge
  3 packages not processed

AndresCao, 2023-02-09:
[image]

Serafadam, 2023-02-10:
Hi, it seems you're not on `foxy` branch, API for vision_msgs changed between different distributions.

fjahncke, 2023-02-13:
you can try 
"sudo apt install ros-foxy-depthai-ros"
"source /opt/ros/foxy/setup.bash"
"ros2 launch depthai_examples rgb_publisher.launch.py"
then you dont have to install from source

AndresCao, 2023-02-14:
> you can try "sudo apt install ros-foxy-depthai-ros" "source /opt/ros/foxy/setup.bash" "ros2 launch depthai_examples rgb_publisher.launch.py" then you dont have to install from source

Thanks, I wasn't aware you added apt install. Thanks!

closed 2023-02-14 by AndresCao