luxonis/depthai-ros #231: [Feature-Request] {Add Nodes/Nodeletsfor detection/segmentation visualization}

Serafadam, 2023-02-17:
### Start with the `why`:
To mimic the experience from base depthai libraries and better visualize detection/segmentation data.

### Move to the `what`:
Currently data streams (images/detections) are decoupled. Adding few small nodes(or nodelets) will add better usability.

### Move to the `how`:
Creating nodelets and using message filters for multiple messages is pretty straightforward, as can be seen in this wls example https://github.com/luxonis/depthai-ros/blob/wls_filter/depthai_ros_driver/src/nodelets/wls_filter.cpp, overlaying detection information/segmentation can be taken from examples in depthai-core. I'm currently only wondering if we should add a separate `depthai_filters` package for them, not to pollute driver code. @saching13

saching13, 2023-02-17:
like the idea of sepearte package named `depthai_filters`!!

Serafadam, 2023-03-27:
Added in https://github.com/luxonis/depthai-ros/pull/247

closed 2023-03-27 by Serafadam