luxonis/depthai-ros #232: [BUG] {Catkin build}

yifan96, 2023-02-21:
I used catkin build with ROS noetic. It worked fine on my laptop but poped errors on intel NUC pc.
Anyone had similar issue before? Thanks in advance.

shafter3d@shafter3d-NUC11TNHi7:depthai_ws$ catkin build 
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/shafter3d/catkin_workspaces/depthai_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/shafter3d/catkin_workspaces/depthai_ws/build
Devel Space:        [exists] /home/shafter3d/catkin_workspaces/depthai_ws/devel
Install Space:      [unused] /home/shafter3d/catkin_workspaces/depthai_ws/install
Log Space:          [exists] /home/shafter3d/catkin_workspaces/depthai_ws/logs
Source Space:       [exists] /home/shafter3d/catkin_workspaces/depthai_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] Found 5 packages in 0.0 seconds.                                       
[build] Package table is up to date.                                           
Starting  >>> depthai_ros_msgs                                                 
Finished  <<< depthai_ros_msgs                  [ 0.2 seconds ]                
Starting  >>> depthai_bridge                                                   
Finished  <<< depthai_bridge                    [ 0.2 seconds ]                
Starting  >>> depthai_examples                                                 
Starting  >>> depthai_ros_driver                                               
Finished  <<< depthai_examples                  [ 0.4 seconds ]                
_______________________________________________________________________________
Errors     << depthai_ros_driver:make /home/shafter3d/catkin_workspaces/depthai_ws/logs/depthai_ros_driver/build.make.004.log
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/src/camera.cpp: In member function ‘void depthai_ros_driver::Camera::getDeviceType()’:
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/src/camera.cpp:72:25: error: ‘using element_type = class dai::Device’ {aka ‘class dai::Device’} has no member named ‘getDeviceName’; did you mean ‘getDeviceInfo’?
   72 |     auto name = device->getDeviceName();
      |                         ^~~~~~~~~~~~~
      |                         getDeviceInfo
In file included from /home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/src/dai_nodes/sensors/sensor_helpers.cpp:1:
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/dai_nodes/sensors/sensor_helpers.hpp: In member function ‘void depthai_ros_driver::dai_nodes::sensor_helpers::ImageSensor::getSizeFromResolution(const dai::ColorCameraProperties::SensorResolution&, int&, int&)’:
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/dai_nodes/sensors/sensor_helpers.hpp:48:64: error: ‘THE_1200_P’ is not a member of ‘dai::ColorCameraProperties::SensorResolution’; did you mean ‘THE_1080_P’?
   48 |             case dai::ColorCameraProperties::SensorResolution::THE_1200_P: {
      |                                                                ^~~~~~~~~~
      |                                                                THE_1080_P
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/dai_nodes/sensors/sensor_helpers.hpp:53:64: error: ‘THE_5_MP’ is not a member of ‘dai::ColorCameraProperties::SensorResolution’; did you mean ‘THE_12_MP’?
   53 |             case dai::ColorCameraProperties::SensorResolution::THE_5_MP: {
      |                                                                ^~~~~~~~
      |                                                                THE_12_MP
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/dai_nodes/sensors/sensor_helpers.hpp:63:64: error: ‘THE_4000X3000’ is not a member of ‘dai::ColorCameraProperties::SensorResolution’
   63 |             case dai::ColorCameraProperties::SensorResolution::THE_4000X3000: {
      |                                                                ^~~~~~~~~~~~~
/home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/dai_nodes/sensors/sensor_helpers.hpp:68:64: error: ‘THE_5312X6000’ is not a member of ‘dai::ColorCameraProperties::SensorResolution
   73 |             case dai::ColorCameraProperties::SensorResolution::THE_48_MP: {
      |                                                                ^~~~~~~~~
      |                                                                THE_12_MP
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/string:41,
                 from /opt/ros/noetic/include/depthai/utility/Initialization.hpp:3,
                 from /opt/ros/noetic/include/depthai/depthai.hpp:6,
                 from /home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/include/depthai_ros_driver/camera.hpp:3,
                 from /home/shafter3d/catkin_workspaces/depthai_ws/src/depthai-ros/depthai_ros_driver/src/camera.cpp:1:
[... 498 lines cut ...]
/opt/ros/noetic/include/depthai/device/Device.hpp:29:37: note:   no known conversion for argument 1 from ‘const dai::DeviceInfo’ to ‘const dai::Pipeline&’
   29 |     explicit Device(const Pipeline& pipeline);
      |                     ~~~~~~~~~~~~~~~~^~~~~~~~
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:63: CMakeFiles/depthai_ros_driver.dir/src/camera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:141: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/sensor_helpers.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:76: CMakeFiles/depthai_ros_driver.dir/src/pipeline_generator.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:115: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/mono.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:154: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/stereo.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:128: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/camera_sensor.cpp.o] Error 1
make[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:102: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/rgb.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:713: CMakeFiles/depthai_ros_driver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/shafter3d/catkin_workspaces/depthai_ws/build/depthai_ros_driver; catkin build --get-env depthai_ros_driver | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << depthai_ros_driver:make           [ Exited with code 2 ]         
Failed    <<< depthai_ros_driver                [ 15.3 seconds ]               
Abandoned <<< depthai-ros                       [ Unrelated job failed ]       
[build] Summary: 3 of 5 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 15.7 seconds total.

Serafadam, 2023-02-22:
Hi, did you install `depthai` library via rosdep (or apt), or did you build it locally?

yifan96, 2023-02-23:
> 

I installed it with rosdep

Serafadam, 2023-02-28:
Could you try running `sudo apt upgrade` and then try building it one more time?

closed 2023-03-27 by Serafadam