luxonis/depthai-ros #233: [BUG] Luxonis Oak 1 POE Examples don't work

iliabaranov, 2023-02-22:
**Check if issue already exists**
- Did a bunch of searching, can't find it :(

**Describe the bug**
I cannot get any of the existing examples of the code to work with the Oak 1 POE cameras, there doesn't seem to be a way to access them. I know it's working, as I've gotten:
IP locked to: 192.168.0.10
gotten the ID: 1844301071AD100900

However, there doesn't seem to be a way to launch an example ROS node to show camera data (just RGB :) ) with the above ID or IP address. Running default examples returns no cameras found, unclear why.

Any examples or suggestions? 
(Sorry, not sure if this is a bug, or a feature that doesn't exist?)

Serafadam, 2023-02-22:
Hi, which executables have you tried?

iliabaranov, 2023-02-22:
Hey, thanks for the reply @Serafadam !

Sorry, I should have mentioned I'm using ROS Noetic.

I've tried:
https://github.com/luxonis/depthai-ros/blob/noetic/depthai_ros_driver/launch/camera.launch
With camera_model:=OAK-POE

and also:
https://github.com/luxonis/depthai-ros/blob/noetic/depthai_examples/launch/mobile_publisher.launch
With camera_model:=OAK-POE

iliabaranov, 2023-02-22:
More info:

When I run the above, I get different variations on:
No devices detected!

However, I can run the following python code:
```python
import depthai

pipeline = depthai.Pipeline()

print ("List of Devices")
for device in depthai.Device.getAllAvailableDevices():
    print(f"{device.getMxId()} {device.state}")

for device in depthai.DeviceBootloader.getAllAvailableDevices():
    print(f"{device.getMxId()} {device.state}")

device_info = depthai.DeviceInfo("192.168.0.10") # IP Address
print (device_info)

with depthai.Device(pipeline,device_info) as device:
  # Print MxID, USB speed, and available cameras on the device
  print('MxId:',device.getDeviceInfo().getMxId())
```

However, the output I get shows that find devices fails, and only explicitly calling out the IP works:

```bash
python3 find_cameras.py 
List of Devices
DeviceInfo(name=192.168.0.10, mxid=, X_LINK_ANY_STATE, X_LINK_ANY_PROTOCOL, X_LINK_ANY_PLATFORM, X_LINK_SUCCESS)
[2023-02-22 11:29:13.835] [warning] Monitor thread (device: 1844301071AD100900 [192.168.0.10]) - ping was missed, closing the device connection
MxId: 1844301071AD100900

```

iliabaranov, 2023-02-22:
I notice with my oak D that it gets in a bad state, and I need to power cycle it to get it to respond again.
On a remote piece of equipment, we can't really do this. Any way around this?

iliabaranov, 2023-02-22:
with camera.launch, I changed the yaml to have the ip address at 192.168.0.10
and I no longer get the line during launch:
"[ INFO] [1677095139.791545410]: No ip/mxid/usb_id specified, connecting to the next available device."
So it does seem to be paying attention to the IP set, but just doesn't seem to connect to it

Serafadam, 2023-02-23:
Can you connect to it and display data from [the demo script](https://docs.luxonis.com/en/latest/#demo-script)?

iliabaranov, 2023-02-24:
Ok, I finally figured it out! 
We were running a firewall that was blocking connection for video streams to the cameras, but was not blocking the initial connection request.

Closing this as a non-issue, but good for people to know to disable their firewall on computer/router during initial testing.

Thanks for the support @Serafadam !

closed 2023-02-24 by iliabaranov

Serafadam, 2023-02-27:
Thanks for the update :slightly_smiling_face: