luxonis/depthai-ros #234: [custom] Can we use nnwrapper to run a custom NN on oak pro d ?

r4hul77, 2023-02-23:


Serafadam, 2023-02-23:
Hi, that depends on what type of NN do you want. Currently, for basic NN we support detection/segmentation, but we'll be adding raw NN output as well as other types in the future

r4hul77, 2023-03-01:
Is there any way I can hack around and get to this. I'm planning to use CNN with an output of 2 floats as an output for my thesis. If I can get this to work It would be a life saver

Serafadam, 2023-03-03:
I'll try adding this option next week. If you're in a hurry, here are the steps to set it up the simplest way. 
- You'd have to add another class in `dai_nodes` directory (let's say `raw_nn.cpp/hpp`), you can use `segmentation` node as a blueprint for that, main change would be to change publishers from image transport to [FloatArray](http://docs.ros.org/en/melodic/api/std_msgs/html/index-msg.html) and custom callback function for that. 
- Then make a config similar [to this one](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/nn/segmentation.json), where you need to change `NN_family` to `raw` (This name might change, but it doesn't matter for now) and update model path.
- Then update functions [here](https://github.com/luxonis/depthai-ros/blob/49152be9afcbfdb5749a43ff9531a8a793c55917/depthai_ros_driver/src/param_handlers/nn_param_handler.cpp#L21)  and [here](https://github.com/luxonis/depthai-ros/blob/49152be9afcbfdb5749a43ff9531a8a793c55917/depthai_ros_driver/src/dai_nodes/nn/nn_wrapper.cpp#L16) to return proper class. After that, build the code, pass path to config file in parameters and run.
If you have any further questions please ask.

closed 2023-09-11 by Serafadam