luxonis/depthai-ros #236: [Feature-Request] Move stl files to separate description package

Aposhian, 2023-02-23:
### Start with the `why`:
stl files take up a lot of space (~60MiB)

### Move to the `what`:
create new package: `depthai_description` that holds urdf and stl files

### Move to the `how`:

Serafadam, 2023-02-27:
Hi, I agree with putting those files in another package for clarity, but size of the package will remain the same in that case (unless I'm missing something)

Aposhian, 2023-02-27:
You don't need the URDF of the oak-d's in order to make use of features like the bridge publisher, so if you break it out and make `depthai_bridge` not depend on `depthai_description` then you can install `depthai_bridge` but not necessarily `depthai_description`. Then that decreases your total install load.

Serafadam, 2023-02-28:
Makes sense :+1:

Serafadam, 2023-03-27:
Added in https://github.com/luxonis/depthai-ros/pull/256

closed 2023-03-27 by Serafadam