luxonis/depthai-ros #237: [Feature-Request] {PointCloud to LaserScan on device}

modulab, 2023-02-23:
I would like to suggest a new feature like the pointcloud_to_laserscan ROS package that would allow it to process data from an on-board camera device by performing the necessary computations on the camera itself. Specifically, the package should be able to receive depth map information directly from the camera and convert it into a pointcloud format. Additionally, the package should apply the necessary 3D transformations to account for the actual camera position, and then flatten the 3D pointcloud into a 2D laser_scan topic that can be used for further processing.

Performing the processing on the camera itself would greatly enhance the efficiency of the system by reducing the amount of data that needs to be transferred. This would be especially beneficial for systems with limited network bandwidth or processing power.

I believe that this feature would greatly enhance the usability and efficiency of the pointcloud_to_laserscan package, and would be a valuable addition for many users. 

That being said, I have no idea if this is possible or where to start to asses it. So if you can please add any comment or advice I would apreciate it. 

Thank you for considering this feature request.

Serafadam, 2023-03-06:
Hi, we've discussed it and although it's not on near-future feature list it could be moved up depending on user demand, for now I'd suggest using either `depthimage_to_laserscan` package or `laserscan_kinect` from [depth_nav_tools](https://github.com/mdrwiega/depth_nav_tools). If we follow up with the development, I'll make sure to tag you here.

EndlessLoops, 2023-03-07:
I used the `depthimage_to_laserscan` package on tb4 but I found it converts poor scan data compared to d435i camera.I'm not sure if the problem is in the parameter configuration. Have you ever had a test like that?

> Hi, we've discussed it and although it's not on near-future feature list it could be moved up depending on user demand, for now I'd suggest using either `depthimage_to_laserscan` package or [...]

modulab, 2023-03-07:
@Serafadam Thanks for answering my call. I'm fairly sure that if such feature would be of much use for those with SBCs and lower end computers, us included. As I think of it, the most resource intense part would be the transform of the point cloud, just before collapsing it into a 2D profile. This would take much the load of the main CPU if done on the camera itself. 
Can you spare some time to steer me in the right direction if i were to tackle this task?

Serafadam, 2023-03-09:
> I used the depthimage_to_laserscan package on tb4 but I found it converts poor scan data compared to d435i camera.I'm not sure if the problem is in the parameter configuration. Have you ever had a test like that?
I expect that playing with stereo depth parameters can have impact here, although we didn't do tests for it (yet). 

> @Serafadam Thanks for answering my call. I'm fairly sure that if such feature would be of much use for those with SBCs and lower end computers, us included. As I think of it, the most resource [...]

Until we implement it in firmware, what I could suggest to try out is [implementing this example](https://docs.luxonis.com/en/latest/pages/tutorials/device-pointcloud/) using `depthai_bridge`, similar to other example nodes in `depthai_examples`. Additional reference [here](https://docs.luxonis.com/en/latest/pages/tutorials/creating-custom-nn-models/#run-your-own-cv-functions-on-device)

Erol444, 2023-03-30:
@modulab @EndlessLoops @Serafadam fwiw we will be adding Pointcloud node to depthai, so you'd be able to gneerate pointcloud on the OAK camera itself. [More information here](https://github.com/luxonis/depthai-core/issues/791).

Erol444, 2023-06-08:
@modulab I believe it's currently WIP - best to pitch in on the feature req. linked above:)