luxonis/depthai-ros #244: [custom] OAK-1 auto-focus mode settings

yifan96, 2023-03-02:
Hi,
I'm trying to mount the OAK-1 auto focus camera on a drone as a down-looking camera to track QR code for landing. The FOV of the OAK-1 is pretty good; however,  the image seems quite blurry due to violent shaking of the drone. Can you suggest the setting mode of the camera? If it should be fixed focus, can you suggest the focus length (a value from 0-255)? 
Thanks.

saching13, 2023-03-03:
We do sell fixed focus version due to this issue on the drones.

yifan96, 2023-03-06:
thanks for your reply.
I have already bought the auto focus camera and would like to mount it on the drone. Is there a way to set its focal length so that auto-focus one can have a better performance on 00000000000drones

Serafadam, 2023-03-08:
Hi, I would say that you have to test it out empirically, you can set manual focus manually in `rqt` using ROS parameters (you can see an example in one of the gifs in our Readme) and then put that value into camera configuration file

saching13, 2023-03-08:
> 

@yifan96  when did you buy this ? Fixing to manual focus might not fix the issue IIRC.

yifan96, 2023-03-08:
Hi, I bought it few weeks before. I did not know which mode suits best for drone QR code tracking and landing. Thought I can manual set focal length for the auto-focus one so I just bought auto-focus cam.

saching13, 2023-03-08:
Gotcha. shoot us a mail if setting manual focus doesn't work. We will see what we can do.

closed 2023-09-11 by Serafadam