luxonis/depthai-ros #245: [BUG] {OAK-D Lite Camera not working}

Grigor-Pahlevanyan, 2023-03-02:
Hi, 

I am using the OAK-D-LITE camera on NVIDIA Jetson Nano single board computer. I am using ROS Noetic on Ubuntu 20.04.5 LTS. I've went through the setup procedure (steps 1 - 8) successfully, but the camera doesn't turn on for all the programs that I run. 

Below are some screenshots. 

[image]
[image]
[image]
[image]

Please let me know what needs to be done to resolve this problem. Thank you.

Serafadam, 2023-03-03:
Hi, can you run the [demo script](https://docs.luxonis.com/en/latest/#demo-script) without issues?

Grigor-Pahlevanyan, 2023-03-03:
Hi, yes the demo works. There is a small numpy & wheel version error but that didn't stop it from working. 

[image]

Grigor-Pahlevanyan, 2023-03-07:
@Serafadam would you please help resolve this issue?

Grigor-Pahlevanyan, 2023-03-07:
I accedentally closed and re-opened, but the issue is still not fixed, so I would appreciate any help.

Grigor-Pahlevanyan, 2023-03-15:
@Serafadam The repo is on the latest noetic branch. I just tried that fist command that you mentioned but still failed. I installed it using the steps under "Install from Source" I didn't use the docker, not sure if that makes a difference. When I ran the last command in your response message I got the following in my terminal:

jetson@nano:~/dai_ws/src/depthai-ros/depthai_ros_driver/launch$ export DEPTHAI_DEBUG=1
jetson@nano:~/dai_ws/src/depthai-ros/depthai_ros_driver/launch$ roslaunch calibration.launch 
... logging to /home/jetson/.ros/log/09af049e-c36f-11ed-b2eb-834c74ef89e8/roslaunch-nano-13960.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nano:43309/

SUMMARY
========

PARAMETERS
 * /oak/camera_i_nn_type: none
 * /oak/camera_i_pipeline_type: RGBStereo
 * /oak/left_i_publish_topic: True
 * /oak/right_i_publish_topic: True
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    calibrator (camera_calibration/cameracalibrator.py)
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [13996]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 09af049e-c36f-11ed-b2eb-834c74ef89e8
process[rosout-1]: started with pid [14012]
started core service [/rosout]
process[oak_nodelet_manager-2]: started with pid [14020]
process[oak-3]: started with pid [14030]
process[calibrator-4]: started with pid [14031]
[ INFO] [1678911721.434533293]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1678911721.472428718]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1678911721.643295819]: Initializing nodelet with 4 worker threads.
[ INFO] [1678911721.674412945]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[2023-03-15 16:22:04.815] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-03-15 16:22:04.815] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-11-28 18:23:08 +0000
[2023-03-15 16:22:04.820] [debug] Initialize - finished
[ INFO] [1678911724.854235420]: No ip/mxid/usb_id specified, connecting to the next available device.
[2023-03-15 16:22:05.088] [debug] Device - OpenVINO version: 2022.1
[2023-03-15 16:22:05.094] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01
[2023-03-15 16:22:05.550] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 18ms, archive read: 709ms
[2023-03-15 16:22:06.991] [debug] Resources - Archive 'depthai-device-fwp-adbcc016c8bd5a5580a26d8b6250f77160203666.tar.xz' open: 12ms, archive read: 2150ms
[2023-03-15 16:22:07.891] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=19443010F1FBF71200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[19443010F1FBF71200] [1.2] [0.832] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304419 should be: 0xFEC083B1
GPIO boot mode 0x16, interface USBD
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0006
Reading VERSION 6 --- ->
eeprom configuration load from user area, status: 0
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: ff ff ff
Invalid Flash JEDEC ID... No NOR available
Could not register the spi device
[... 184 lines cut ...]
[19443010F1FBF71200] [1.2] [9.846] [system] [info] Temperatures - Average: 34.82 °C, CSS: 36.53 °C, MSS 34.64 °C, UPA: 33.93 °C, DSS: 34.16 °C
[19443010F1FBF71200] [1.2] [9.846] [system] [info] Cpu Usage - LeonOS 21.75%, LeonRT: 1.32%
[19443010F1FBF71200] [1.2] [10.847] [system] [info] Memory Usage - DDR: 135.53 / 340.61 MiB, CMX: 2.37 / 2.50 MiB, LeonOS Heap: 49.39 / 77.48 MiB, LeonRT Heap: 4.14 / 41.35 MiB
[19443010F1FBF71200] [1.2] [10.847] [system] [info] Temperatures - Average: 34.82 °C, CSS: 35.59 °C, MSS 34.64 °C, UPA: 34.64 °C, DSS: 34.40 °C
[19443010F1FBF71200] [1.2] [10.847] [system] [info] Cpu Usage - LeonOS 21.93%, LeonRT: 1.33%
[19443010F1FBF71200] [1.2] [11.848] [system] [info] Memory Usage - DDR: 135.53 / 340.61 MiB, CMX: 2.37 / 2.50 MiB, LeonOS Heap: 49.39 / 77.48 MiB, LeonRT Heap: 4.14 / 41.35 MiB
[19443010F1FBF71200] [1.2] [11.848] [system] [info] Temperatures - Average: 35.29 °C, CSS: 36.53 °C, MSS 35.11 °C, UPA: 34.64 °C, DSS: 34.88 °C
[19443010F1FBF71200] [1.2] [11.848] [system] [info] Cpu Usage - LeonOS 21.77%, LeonRT: 1.32%
[19443010F1FBF71200] [1.2] [12.849] [system] [info] Memory Usage - DDR: 135.53 / 340.61 MiB, CMX: 2.37 / 2.50 MiB, LeonOS Heap: 49.39 / 77.48 MiB, LeonRT Heap: 4.14 / 41.35 MiB
[... cut ...]
[19443010F1FBF71200] [1.2] [13.850] [system] [info] Temperatures - Average: 34.82 °C, CSS: 36.06 °C, MSS 34.64 °C, UPA: 34.16 °C, DSS: 34.40 °C
[19443010F1FBF71200] [1.2] [13.850] [system] [info] Cpu Usage - LeonOS 21.68%, LeonRT: 1.33%

Here is how my bashrc looks like in case there is anything wrong in here:
[image]

Here is the overall setup of the directories.
[image]

Please let me know what you think may be the issue. I can try deleting everything and re-doing but don't think that would make much of a difference. Would it be possible to schedule a meeting with you if I am still unable to resolve the issue?

Thanks,
Grigor

Serafadam, 2023-03-16:
Hi Grigor,
Hmm, according to those logs everything should be fine :thinking: 
I'll setup my Nano today/tomorrow and check it out, until that some additional quick questions:
1. When running nodes, are you able to see images using rqt/rqt_image_view? Or can you echo them (for example `rostopic echo /oak/rgb/image_raw`?
2. Can you publish some random image with [image_publishe](http://wiki.ros.org/image_publisher)` and display it in rviz or rqt?
3. Is this a fresh ROS/System install? Are you working with some other packages here and do they work without problems?
4. Could you try running `docker run -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev:/dev --runtime nvidia --network host luxonis/depthai-ros:noetic-latest roslaunch depthai_ros_driver camera.launch` and see if it works using prebuilt image?

Grigor-Pahlevanyan, 2023-03-17:
Hi @Serafadam 

1. I am not able to see image using rqt/rqt_image_view. Here are some screenshots.
[image]
[image]

1.b. Using echo doesn't output any data either. So its clear that the data is missing. 
[image]

2. Random image publishing with rqt does work. Below is a screenshot. 
[image]

3. This is a fairly fresh install of ROS (installed it 2 months ago). I started learning ROS recently and have created packages myself that didn't involve any cameras. 

4. Here is what I get by running `docker run -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev:/dev --runtime nvidia --network host luxonis/depthai-ros:noetic-latest roslaunch depthai_ros_driver camera.launch`. So pretty much running it with the docker made no difference, if I launch camera.launch manually like roslaunch camera.launch it fails the same way.

```
jetson@nano:~$ sudo docker run -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev:/dev --runtime nvidia --network host luxonis/depthai-ros:noetic-latest roslaunch depthai_ros_driver camera.launch
Unable to find image 'luxonis/depthai-ros:noetic-latest' locally
noetic-latest: Pulling from luxonis/depthai-ros
70cf24b16239: Already exists 
69c8f9dc3300: Already exists 
f3873cd7c37c: Already exists 
11990ca89946: Already exists 
b170ac8cf943: Already exists 
a80cc2bb1bc5: Already exists 
ae0dc1f7d500: Already exists 
7f807d9f223f: Already exists 
000cc58540f3: Already exists 
8ceb784b7cdf: Already exists 
daa96515c599: Pull complete 
180c38a74734: Pull complete 
d939d422db96: Pull complete 
15163b2694f7: Pull complete 
0dd83b25972c: Pull complete 
269f117ab5b1: Pull complete 
1b1df3fed04b: Pull complete 
d32a67a957d2: Pull complete 
098cdcae5016: Pull complete 
Digest: sha256:a6a536ae5e82ecff94e3de23ff1873f4518624c4ec755a941b3e7b376d7f8e3b
Status: Downloaded newer image for luxonis/depthai-ros:noetic-latest
... logging to /root/.ros/log/4eb40258-c462-11ed-b70b-3f4138b53589/roslaunch-nano-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nano:38601/

SUMMARY
========

PARAMETERS
 * /oak/camera_i_enable_imu: True
 * /oak/camera_i_enable_ir: True
 * /oak/camera_i_floodlight_brightness: 0.0
 * /oak/camera_i_ip: 
 * /oak/camera_i_laser_dot_brightness: 800.0
 * /oak/camera_i_mx_id: 
 * /oak/camera_i_nn_type: spatial
 * /oak/camera_i_pipeline_type: RGBD
 * /oak/camera_i_usb_port_id: 
 * /oak/camera_i_usb_speed: SUPER_PLUS
 * /oak/imu_i_max_q_size: 30
 * /oak/left_i_board_socket_id: 1
 * /oak/left_i_calibration_file: 
 * /oak/left_i_fps: 30.0
 * /oak/left_i_height: 720
 * /oak/left_i_low_bandwidth: False
 * /oak/left_i_low_bandwidth_quality: 50
 * /oak/left_i_max_q_size: 30
[... 137 lines cut ...]
[ INFO] [1679017012.305862126]: Camera with MXID: 19443010F1FBF71200 and Name: 1.2 connected!
[ INFO] [1679017012.307575446]: USB SPEED: SUPER
[ INFO] [1679017012.343307249]: Device type: OAK-D-LITE
[ WARN] [1679017012.346836184]: Device has no IR drivers
[ INFO] [1679017012.356968602]: Pipeline type: RGBD
[ INFO] [1679017012.630163781]: NN Family: mobilenet
[ INFO] [1679017013.034977321]: Finished setting up pipeline.
[19443010F1FBF71200] [1.2] [1.759] [MonoCamera(4)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[19443010F1FBF71200] [1.2] [1.759] [MonoCamera(6)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[FATAL] [1679017016.462600271]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-2] process has died [pid 53, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/root/.ros/log/4eb40258-c462-11ed-b70b-3f4138b53589/oak_nodelet_manager-2.log].
log file: /root/.ros/log/4eb40258-c462-11ed-b70b-3f4138b53589/oak_nodelet_manager-2*.log
================================================================================REQUIRED process [oak-3] has died!
process has died [pid 54, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load depthai_ros_driver/Camera oak_nodelet_manager __name:=oak __log:=/root/.ros/log/4eb40258-c462-11ed-b70b-3f4138b53589/oak-3.log].
log file: /root/.ros/log/4eb40258-c462-11ed-b70b-3f4138b53589/oak-3*.log
Initiating shutdown!
================================================================================
[oak-3] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
jetson@nano:~$ 
```
Please let me know if you are able to figure out the issue with your Jetson Nano. If not do let me know I am open for a meeting.

Serafadam, 2023-03-17:
I was able to get it working on my Nano (Ubuntu 18.04, I used docker to mimic fresh setup on 20.04), so it's not a general issue. Other ideas that I have:
1. Can you see anything on `/oak/rgb/image_raw/compressed` topic?
2. Are you using [power adapter](https://shop.luxonis.com/collections/accessories/products/oak-y-adapter)? 
3. Are your apt packages up-to-date? 
4. Maybe for some reason some crucial package has not been installed, you can try running `sudo curl -fL https://docs.luxonis.com/install_dependencies.sh | bash`
5. Let's try stereo_inertial_node in docker, run:
`xhost +local:docker`
then
`docker run -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev:/dev --runtime nvidia --network host luxonis/depthai-ros:noetic-latest roslaunch depthai_examples stereo_inertial_node.launch`
6. If it still doesn't work, I would try reinstalling the system and trying it out on a fresh one (maybe starting with docker approach) if that's not a problem. I'll also consult my colleagues on that issue and post if there are any additional ideas, we can set up a meeting, but after trying steps above as we'll have some additional information.

Grigor-Pahlevanyan, 2023-03-17:
@Serafadam I am going to start completely from scratch and see if it works this time around. The options you've mentioned above still failed. I am not using a power adapter by the way. Its connected by 1 USB C which deals both with power and data transfer. It is also connected to the USB 3 port so that shouldn't be the problem either. The camera image does work on the demo that I tried earlier so that would eliminate the connector from being the problem. 

Where did you get your Ubuntu image from for the Jetson. I got mine from:
 https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image

Maybe that could be the issue, I am going to get a new image of Ubuntu 20.04 from the official website. If that fails, I might try again from the NVIDIA website, I'm just not sure if this is ubuntu or what version it is.
https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write

Please let me know if which one do you think would be the better one to go with. I will contact you once both options fail. 

Thanks.

saching13, 2023-03-17:
@Grigor-Pahlevanyan  can you prefix the execution commands with `DEPTHAI_LEVEL=debug` and share the logs.

Grigor-Pahlevanyan, 2023-03-17:
@saching13 I've already deleted the OS. Will re insatll it and if it still fails will try that.

MAVProxyUser, 2023-03-18:
"noetic" eek! That is old. Bump your jetpack version, hand compile humble. And use newer branches/ 
From a fresh 5.1 Jetpack install 
```
sudo apt-get remove python2.7 libpython2.7-dev python2-minimal python2.7-minimal     libpython2.7 
sudo apt-get remove libpython3.9-minimal libpython3.9-stdlib python3.9 python3.9-minimal python3.9-vpi2
sudo apt-get install python-is-python3

https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html
sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash
wget https://github.com/Kitware/CMake/releases/download/v3.26.0-rc1/cmake-3.26.0-rc1.tar.gz
tar -zxvf cmake-3.26.0-rc1.tar.gz
cd cmake-3.26.0-rc1/
./configure ; sudo make install
```
In the luxonis ros workspace, ensure you have:
```
ubuntu@ubuntu:~/luxonis_oak_ws/src$ git clone https://github.com/ros-perception/vision_msgs.git
ubuntu@ubuntu:~/luxonis_oak_ws/src$ git clone https://github.com/ros-perception/vision_opencv.git -b humble
ubuntu@ubuntu:~/luxonis_oak_ws/src$ git clone https://github.com/ros/xacro.git -b ros2
ubuntu@ubuntu:~/luxonis_oak_ws/src$ git clone https://github.com/ros-perception/image_pipeline.git -b humble
```
And colcon build the luxonis code.

MAVProxyUser, 2023-03-18:
> Hi, yes the demo works. There is a small numpy & wheel version error but that didn't stop it from working.
> 
> [image]

From my experience on OSX, this is not what the Demo looks like. This is a partial demo window.

Grigor-Pahlevanyan, 2023-03-20:
Updates:
1. I have installed Ubuntu 18.04, but that doesn't support ROS Noetic. So instead of using a docker I tried Ubuntu 20.04.
2. @Serafadam Installed Ubuntu 20.04 from https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image which is what I had initially. Ubuntu 20.04 from the official Ubuntu website doesn't load on the Jetson. By the way, mine is a Jetson Nano 2GB not 4GB not sure if that could be an issue. 
3. Here are more logs 

[image]

Running catkin_make (may help in debugging)
```
jetson@nano:~/dai_ws$ catkin_make_isolated -l1 -j1
Base path: /home/jetson/dai_ws
Source space: /home/jetson/dai_ws/src
Build space: /home/jetson/dai_ws/build_isolated
Devel space: /home/jetson/dai_ws/devel_isolated
Install space: /home/jetson/dai_ws/install_isolated
Additional make Arguments: -l1 -j1
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - depthai-ros
~~  - depthai_ros_msgs
~~  - depthai_bridge
~~  - depthai_examples
~~  - depthai_ros_driver
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'depthai-ros'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/jetson/dai_ws/build_isolated/depthai-ros'
==> make -l1 -j1 in '/home/jetson/dai_ws/build_isolated/depthai-ros'
<== Finished processing package [1 of 5]: 'depthai-ros'

==> Processing catkin package: 'depthai_ros_msgs'
==> Building with env: '/home/jetson/dai_ws/devel_isolated/depthai-ros/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/jetson/dai_ws/build_isolated/depthai_ros_msgs'
==> make -l1 -j1 in '/home/jetson/dai_ws/build_isolated/depthai_ros_msgs'
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_HandLandmarkArray
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_NormalizedImageCrop
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_SpatialDetection
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_AutoFocusCtrl
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_SpatialDetectionArray
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_TriggerNamed
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target _depthai_ros_msgs_generate_messages_check_deps_HandLandmark
[  0%] Built target vision_msgs_generate_messages_cpp
[ 18%] Built target depthai_ros_msgs_generate_messages_cpp
[ 18%] Built target sensor_msgs_generate_messages_py
[ 18%] Built target geometry_msgs_generate_messages_py
[ 18%] Built target std_msgs_generate_messages_py
[ 18%] Built target vision_msgs_generate_messages_py
[ 42%] Built target depthai_ros_msgs_generate_messages_py
[ 42%] Built target geometry_msgs_generate_messages_nodejs
[ 42%] Built target sensor_msgs_generate_messages_nodejs
[ 42%] Built target std_msgs_generate_messages_nodejs
[ 42%] Built target vision_msgs_generate_messages_nodejs
[ 60%] Built target depthai_ros_msgs_generate_messages_nodejs
[ 60%] Built target geometry_msgs_generate_messages_lisp
[ 60%] Built target sensor_msgs_generate_messages_lisp
[ 60%] Built target std_msgs_generate_messages_lisp
[ 60%] Built target vision_msgs_generate_messages_lisp
[ 78%] Built target depthai_ros_msgs_generate_messages_lisp
[ 78%] Built target geometry_msgs_generate_messages_eus
[ 78%] Built target vision_msgs_generate_messages_eus
[ 78%] Built target sensor_msgs_generate_messages_eus
[ 78%] Built target std_msgs_generate_messages_eus
[100%] Built target depthai_ros_msgs_generate_messages_eus
[100%] Built target depthai_ros_msgs_generate_messages
<== Finished processing package [2 of 5]: 'depthai_ros_msgs'
[... 1119 lines cut ...]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]
[Processing: depthai_examples]

Summary: 3 packages finished [57min 13s]
  1 package failed: depthai_examples
  1 package had stderr output: depthai_examples
  1 package not processed
The command '/bin/sh -c cd .$WS/ && . /opt/ros/${ROS_DISTRO}/setup.sh && ./src/depthai-ros/build.sh -s $BUILD_SEQUENTIAL -r 1 -m 1' returned a non-zero code: 2
jetson@nano:~/catkin_ws/src/depthai-ros$ 
```

@MAVProxyUser do you know if there are any tutorials online that I can follow to make those updates. Unfortunately I can't install ROS2 Humble on Ubuntu 20.04 it has to be 22.04 and up. So I am not sure how to "Hand Compile" Humbe. I am still new to this territory so tutorials on how to use colcon build, hand compile humble, and how to setup the new JetPack would really help!

Serafadam, 2023-03-22:
After seeing comment from @MAVProxyUser I must agree that this is part of the demo script, now just to be sure, when you run the demo script, do you get GUI like this?
[image]

MAVProxyUser, 2023-03-23:
Ok @Serafadam I did a little investingation comparing my Linux and Mac... seems by default on OSX QT interface is used. 
I saw in the documentation that I should try: 
```
ubuntu@ubuntu:~/depthai$ python3 depthai_demo.py --guiType qt
```

This will on a Linux machine bring up the full QT demo as seen on OSX. 

[image]

If you don't first install the proper QT modules you won't have success. 
I first saw this error
```
Traceback (most recent call last):
  File "depthai_demo.py", line 1050, in <module>
    runQt()
  File "depthai_demo.py", line 583, in runQt
    from gui.main import DemoQtGui
  File "/home/ubuntu/depthai/gui/main.py", line 7, in <module>
    from PyQt5.QtQml import QQmlApplicationEngine, qmlRegisterType, qmlRegisterSingletonType, QQmlEngine
ModuleNotFoundError: No module named 'PyQt5.QtQml'
Sentry is attempting to send 2 pending error messages
```

I fixed it by installing the following packages
```
ubuntu@ubuntu:~/depthai$ sudo apt-get install python3-pyside2.qtqml 
ubuntu@ubuntu:~/depthai$ sudo apt-get install python3-pyqt5.qtquick python3-pyqt5.qtsql PyQt5.QtOpenGL
```

This in turn lead to more errors when I launched the demo
```
QQmlApplicationEngine failed to load component
file:///home/ubuntu/depthai/gui/views/root.qml:52 module "QtQuick.Layouts" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:53 module "QtQuick.Controls" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:55 module "QtQuick.Controls.Material" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:51 module "QtQuick" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:54 module "QtQuick.Window" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:52 module "QtQuick.Layouts" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:53 module "QtQuick.Controls" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:55 module "QtQuick.Controls.Material" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:51 module "QtQuick" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:54 module "QtQuick.Window" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:52 module "QtQuick.Layouts" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:53 module "QtQuick.Controls" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:55 module "QtQuick.Controls.Material" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:51 module "QtQuick" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:54 module "QtQuick.Window" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:52 module "QtQuick.Layouts" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:53 module "QtQuick.Controls" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:55 module "QtQuick.Controls.Material" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:51 module "QtQuick" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:54 module "QtQuick.Window" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:52 module "QtQuick.Layouts" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:53 module "QtQuick.Controls" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:55 module "QtQuick.Controls.Material" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:51 module "QtQuick" is not installed
file:///home/ubuntu/depthai/gui/views/root.qml:54 module "QtQuick.Window" is not installed
```

I fixed these with a brute install of all the associated qml modules. 
```
ubuntu@ubuntu:~/depthai$ sudo apt-get install "qml-module-qtquicK*"
```

After that everything launched fine with the FULL demo active.

MAVProxyUser, 2023-03-23:
@Grigor-Pahlevanyan regarding "do you know if there are any tutorials online that I can follow to make those updates. Unfortunately I can't install ROS2 Humble on Ubuntu 20.04 it has to be 22.04 and up. So I am not sure how to "Hand Compile" Humbe. I am still new to this territory so tutorials on how to use colcon build, hand compile humble, and how to setup the new JetPack would really help!"

This gets tricky. Exactly which Jetson board are you on? If it is a Nano, things get tricky because your Jetpack version is limited. You can see the product support matrix here: 
https://www.stereolabs.com/blog/nvidia-jetson-l4t-and-jetpack-support/

You are at the mercy of the SDKManager for the most part https://developer.nvidia.com/drive/sdk-manager

I'm on Jetpack 5.1 with a Jetson Xavier NX. I got fed up with outdated Jetson Nano support being locked at JetPack 4.6.1 (and Ubuntu 18 accordingly). 

Generally speaking a "Hand Compile" as I mentioned follows the directions for a "source" compile from the ROS website: https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html (Humble is of course intended for Ubuntuu 20). 

NVIDIA seems to be forcing old Jetson Users on to more expensive platforms in order to get newer linux support. As evidenced by the most recent release the Jetson Orin Nano. https://hackaday.com/2023/03/21/hands-on-nvidia-jetson-orin-nano-developer-kit/

MAVProxyUser, 2023-03-23:
now @Grigor-Pahlevanyan your actual bug resembles something still have when I launch the following:
``` 
ubuntu@ubuntu:~/depthai$ ros2 launch depthai_ros_driver calibration.launch.py camera_model:=OAK-1
```
It doesn't matter if I use my OAK-1-MAX, OAK-1, OAK-D-PRO, or OAK-D-S2. they all in essence close as follows:

```
[component_container-1] UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
[component_container-1] Done!
[component_container-1] Usb connection speed: Super - USB 3.0
[component_container-1] Temperature: Driver registered.
[component_container-1] Temperature: Initialized driver.
[component_container-1] Temperature: Sensor opened: CSS.
[component_container-1] Temperature: Sensor opened: MSS.
[component_container-1] Temperature: Sensor opened: UPA.
[component_container-1] Temperature: Sensor opened: DSS.
[component_container-1] [14442C10A18652D000] [1.2.1] [0.940] [system] [info] Memory Usage - DDR: 0.12 / 340.42 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 7.24 / 77.32 MiB, LeonRT Heap: 2.89 / 41.23 MiB
[component_container-1] [14442C10A18652D000] [1.2.1] [0.940] [system] [info] Temperatures - Average: 31.77 °C, CSS: 32.73 °C, MSS 31.29 °C, UPA: 31.04 °C, DSS: 32.01 °C
[component_container-1] [14442C10A18652D000] [1.2.1] [0.940] [system] [info] Cpu Usage - LeonOS 43.14%, LeonRT: 1.50%
[component_container-1] [14442C10A18652D000] [1.2.1] [0.949] [system] [warning] PRINT:LeonCss: I: [Timesync] [   2127634] [XLin] startSync:130     Timesync | Callback not set
[component_container-1] [INFO] [1679552103.084042340] [oak]: Camera with MXID: 14442C10A18652D000 and Name: 1.2.1 connected!
[component_container-1] [INFO] [1679552103.086671311] [oak]: USB SPEED: SUPER
[component_container-1] [14442C10A18652D000] [1.2.1] [0.971] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] [INFO] [1679552103.101888847] [oak]: Device type: 
[component_container-1] [INFO] [1679552103.110211250] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-23 02:15:03.111] [debug] Device about to be closed...
[component_container-1] [2023-03-23 02:15:03.271] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-23 02:15:03.271] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-23 02:15:03.802] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[component_container-1] [2023-03-23 02:15:04.143] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-23 02:15:04.543] [debug] Device closed, 1430
[component_container-1] [ERROR] [1679552104.562373523] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

But my screen looks identical to yours in that case: 
[image]

I'm tracking mine here where @Serafadam is also helping...
https://github.com/luxonis/depthai-ros/issues/257

These seem related for sure.

Serafadam, 2024-11-12:
Hi, it seems you have a different error, a plugin is missing form image proc, please verify that it is installed - `sudo apt install ros-noetic-image-proc`

hazee007, 2024-11-15:
Alright i did install that, now the error message is lesser
[image]

Serafadam, 2024-11-19:
Hi, thanks for the report, could you follow [those steps](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md) to verify that your camera works properly?