luxonis/depthai-ros #248: [Question] How to set the "maximum frame size specified in properties"?

RemiFabre, 2023-03-11:
Hi,
I'm trying to use a YoloV7 blob with the ROS driver. The blob works fine using a python pipeline, but after increasing the preview_size to 640 for rgb using the ROS driver all the frames are skipped with the following error:

> [component_container-3] [14442C1071A460D700] [3.1] [4.079] [ImageManip(10)] [error] Output image is bigger (1228800B) than maximum frame size specified in properties (1048576B) - skipping frame.

### Versions
Ubuntu 22.04, ROS Humble, OAK-D.

### Problem reproduction
The minimal change to create the issue seems to be:
In camera.yaml set rgb->i_preview_size to 640 (previously was 416)

Then launch:
`ros2 launch depthai_ros_driver camera.launch.py`

### Question
How to set the "maximum frame size specified in properties"? I tried all of the parameters that seemed related but the specified limit in the message never changed.

Note: the error message is coherent with 640 * 640 * 3 = 1228800 Bytes per frame

Best,

Serafadam, 2023-03-13:
Hi, is it possible for you to share the blob?

RemiFabre, 2023-03-14:
Hi,

Thanks for the fast reply.

Sure. If useful, below the initialization section of the Python code that works with the [blob](https://we.tl/t-IGyunuZYre).

```
self.syncNN = True
self.pipeline = dai.Pipeline()
# Define sources and outputs
self.camRgb = self.pipeline.create(dai.node.ColorCamera)
self.spatialDetectionNetwork = self.pipeline.create(
    dai.node.YoloSpatialDetectionNetwork
)
self.monoLeft = self.pipeline.create(dai.node.MonoCamera)
self.monoRight = self.pipeline.create(dai.node.MonoCamera)
self.stereo = self.pipeline.create(dai.node.StereoDepth)
self.nnNetworkOut = self.pipeline.create(dai.node.XLinkOut)

self.xoutRgb = self.pipeline.create(dai.node.XLinkOut)
self.xoutNN = self.pipeline.create(dai.node.XLinkOut)
self.xoutBoundingBoxDepthMapping = self.pipeline.create(dai.node.XLinkOut)
self.xoutDepth = self.pipeline.create(dai.node.XLinkOut)

self.xoutRgb.setStreamName("rgb")
self.xoutNN.setStreamName("detections")
self.xoutBoundingBoxDepthMapping.setStreamName("boundingBoxDepthMapping")
self.xoutDepth.setStreamName("depth")
self.nnNetworkOut.setStreamName("nnNetwork")

# Properties
self.camRgb.setPreviewSize(self.img_size, self.img_size)
self.camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
self.camRgb.setInterleaved(False)
self.camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.BGR)

self.monoLeft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
self.monoLeft.setBoardSocket(dai.CameraBoardSocket.LEFT)
self.monoRight.setResolution(
    dai.MonoCameraProperties.SensorResolution.THE_400_P
)
self.monoRight.setBoardSocket(dai.CameraBoardSocket.RIGHT)

# setting node configs
self.stereo.setDefaultProfilePreset(
    dai.node.StereoDepth.PresetMode.HIGH_DENSITY
)
# Align depth map to the perspective of RGB camera, on which inference is done
self.stereo.setDepthAlign(dai.CameraBoardSocket.RGB)
self.stereo.setOutputSize(
    self.monoLeft.getResolutionWidth(), self.monoLeft.getResolutionHeight()
)

self.spatialDetectionNetwork.setBlobPath(self.nnBlobPath)
self.spatialDetectionNetwork.setConfidenceThreshold(0.5)
self.spatialDetectionNetwork.input.setBlocking(False)
self.spatialDetectionNetwork.setBoundingBoxScaleFactor(0.5)
self.spatialDetectionNetwork.setDepthLowerThreshold(100)
self.spatialDetectionNetwork.setDepthUpperThreshold(5000)

# Yolo specific parameters
self.spatialDetectionNetwork.setNumClasses(80)
self.spatialDetectionNetwork.setCoordinateSize(4)
# yolov4
# self.spatialDetectionNetwork.setAnchors([10, 14, 23, 27, 37, 58, 81, 82, 135, 169, 344, 319])
# self.spatialDetectionNetwork.setAnchorMasks({"side26": [1, 2, 3], "side13": [3, 4, 5]})
# yolov7
self.spatialDetectionNetwork.setAnchors(
    [
        12.0,
        16.0,
        19.0,
        36.0,
        40.0,
        28.0,
        36.0,
        75.0,
        76.0,
        55.0,
        72.0,
        146.0,
        142.0,
        110.0,
        192.0,
        243.0,
        459.0,
        401.0,
    ]
)
self.spatialDetectionNetwork.setAnchorMasks(
    {"side80": [0, 1, 2], "side40": [3, 4, 5], "side20": [6, 7, 8]}
)

self.spatialDetectionNetwork.setIouThreshold(1.0)

# Linking
self.monoLeft.out.link(self.stereo.left)
self.monoRight.out.link(self.stereo.right)

self.camRgb.preview.link(self.spatialDetectionNetwork.input)
if self.syncNN:
    self.spatialDetectionNetwork.passthrough.link(self.xoutRgb.input)
else:
    self.camRgb.preview.link(self.xoutRgb.input)

self.spatialDetectionNetwork.out.link(self.xoutNN.input)
self.spatialDetectionNetwork.boundingBoxMapping.link(
    self.xoutBoundingBoxDepthMapping.input
)

self.stereo.depth.link(self.spatialDetectionNetwork.inputDepth)
self.spatialDetectionNetwork.passthroughDepth.link(self.xoutDepth.input)
self.spatialDetectionNetwork.outNetwork.link(self.nnNetworkOut.input)
```

Serafadam, 2023-03-14:
Ok, thanks for the information, I managed to reproduce this issue and will add a fix later. 
In short - we have ImageManip node that scales RGB preview input to size that fits blob requirements, that way we don't have to edit config files for different blobs that we use, though because of that max size that ImageManip can handle is ~590x590.
To fix that limitation:
- add parameter `nn.i_disable_resize` that omits that step and links preview straight to NN input
- add size checking and throw if size is too large and flag above is not set

For a quick fix, if you have driver built from source:
edit [yolo.cpp](https://github.com/luxonis/depthai-ros/blob/49152be9afcbfdb5749a43ff9531a8a793c55917/depthai_ros_driver/src/dai_nodes/nn/yolo.cpp#L67)

change
```
dai::Node::Input Yolo::getInput(int /*linkType*/) {
    return imageManip->inputImage;
}
```
to
```
dai::Node::Input Yolo::getInput(int /*linkType*/) {
    return yoloNode->input;
}
```

or in case of [spatial_yolo.cpp](https://github.com/luxonis/depthai-ros/blob/49152be9afcbfdb5749a43ff9531a8a793c55917/depthai_ros_driver/src/dai_nodes/nn/spatial_yolo.cpp#L65)
change
```
dai::Node::Input SpatialYolo::getInput(int linkType) {
    if(linkType == static_cast<int>(nn_helpers::link_types::SpatialNNLinkType::input)) {
        return imageManip->inputImage;
    } else {
        return yoloNode->inputDepth;
    }
}
```
to
```
dai::Node::Input SpatialYolo::getInput(int linkType) {
    if(linkType == static_cast<int>(nn_helpers::link_types::SpatialNNLinkType::input)) {
        return yoloNode->input;
    } else {
        return yoloNode->inputDepth;
    }
}
```
and change rgb->i_preview_size to 640
Hope this helps

RemiFabre, 2023-03-14:
Amazing, thanks, we'll try it tomorrow and report back.

RemiFabre, 2023-03-19:
@Serafadam your fix worked wonderfully, thanks :)

A minor feedback, the TFs published for each detection made never die, so eventually you end up with a ton of old TFs:
[image]
[image]

The problem lies in 'obj_pub.py', because the stamp is not defined in the TF header. Adding this line fixes this issue:

`tf.header.stamp = self.get_clock().now().to_msg()`

Best,

closed 2023-09-11 by Serafadam