luxonis/depthai-ros #250: [BUG] {Build error: unknown package type}

22222140, 2023-03-15:
Hello sir.

I'm trying OAK-D Pro on ROS Noetic on my Ubuntu 20.04
And it doesn't work in Setting up procedure.
When running catkin_make_isolated, I can see an error below.

I also tried catkin clean and sudo apt upgrade..

I would appreciate any assistance for it

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - depthai-ros (unknown)
~~  - depthai_ros_msgs (unknown)
~~  - depthai_bridge (unknown)
~~  - depthai_examples (unknown)
~~  - depthai_ros_driver (unknown)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Error: Packages with unknown build types exist
Can not build workspace with packages of unknown build_type

Serafadam, 2023-03-15:
Hi, it seems you need to switch to `noetic` branch

22222140, 2023-03-17:
Hi, Sorry that there was a mistake in my explanation. 
I ran the commands from noetic branch, **catkin_make_isolated -l1 -j1**

One more thing, is it correct to run this command below, after **cd ../..** at **/dai_ws/src** ?
**rosdep install --from-paths src --ignore-src -r -y**
Actually when I do this at ~$, I can see the message that "given path 'src' does not exist"

ceemil, 2023-03-17:
It's a mistake it would be cd .. 
You are pointing rosdep to the src directory for installation.
If you use cd ../.. it should be: rosdep install --from-paths dai_ws/src --ignore-src -r -y

22222140, 2023-04-24:
Thank you, 
Actually I already did cd.. as well but I could see same error message below

ase path: /home/asl/dai_ws
Source space: /home/asl/dai_ws/src
Build space: /home/asl/dai_ws/build_isolated
Devel space: /home/asl/dai_ws/devel_isolated
Install space: /home/asl/dai_ws/install_isolated
Additional make Arguments: -l1 -j1
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 7 packages in topological order:
~~  - depthai-ros (unknown)
~~  - depthai_descriptions (unknown)
~~  - depthai_filters (unknown)
~~  - depthai_ros_msgs (unknown)
~~  - depthai_bridge (unknown)
~~  - depthai_examples (unknown)
~~  - depthai_ros_driver (unknown)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Error: Packages with unknown build types exist
Can not build workspace with packages of unknown build_type

Serafadam, 2023-04-25:
Hi, could you post git branch that you are on?

22222140, 2023-04-26:
I'm trying on noetic branch.