luxonis/depthai-ros #251: Problem with building project using colcon

qnaa42, 2023-03-16:
Hi, I'm pretty new to ROS, and after some battle with settings and packages, I finally came to the building stage.
However, while trying to do so, I'm getting the below error:

colcon build --cmake-clean-cache
Starting >>> depthai_ros_msgs
Finished <<< depthai_ros_msgs [7.32s]
Starting >>> depthai_bridge
Finished <<< depthai_bridge [21.3s]
Starting >>> depthai_examples
Starting >>> depthai_ros_driver
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[7min 10.9s] [2/5 complete] [2 ongoing] [depthai_examples:build 64% - 6min 41.9s] ...[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]xamples:build 64% - 8min 12.3s] ...
[Processing: depthai_examples, depthai_ros_driver]
[10min 47.4s] [2/5 complete] [2 ongoing] [depthai_examples:build 64% - 10min 18.4s] ...[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]examples:build 64% - 11min 41.9s] ...
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver][20min 7.7s] [2/5 complete] [2 ongoing] [depthai_examples:build 64% - 19min 38.7s] ...
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[24min 19.7s] [2/5 complete] [2 ongoing] [depthai_examples:build 64% - 23min 50.8s] ...[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
[Processing: depthai_examples, depthai_ros_driver]
--- stderr: depthai_ros_driver
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:160: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/sensor_helpers.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:286: CMakeFiles/depthai_ros_driver.dir/src/param_handlers/camera_param_handler.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:118: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/rgb.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:174: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/stereo.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:272: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/nn/spatial_yolo.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:244: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/nn/yolo.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:258: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/nn/spatial_mobilenet.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:104: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/imu.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_ros_driver.dir/build.make:132: CMakeFiles/depthai_ros_driver.dir/src/dai_nodes/sensors/mono.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/depthai_ros_driver.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
Failed <<< depthai_ros_driver [27min 36s, exited with code 2]
Aborted <<< depthai_examples [27min 43s]

Summary: 2 packages finished [28min 12s]
1 package failed: depthai_ros_driver
1 package aborted: depthai_examples
2 packages had stderr output: depthai_examples depthai_ros_driver
1 package not processed

ChatGPT is suggesting that it is memory issue after running spawn -s I'm getting
Filename Type Size Used Priority
/dev/sdb partition 2097152 43388 -2

I would greatly appreciate any help or lead you could give me. Im using LWS Ubuntu 22.04 and Windows 11; the purpose is to bind OAK-D with unity to create an art installation for modern art exhibitions using detected people as input for RL agents trained using ML agents.

Serafadam, 2023-03-16:
Hi, it seems that your PC runs out of resources when compiling, this is something that will be improved in https://github.com/luxonis/depthai-ros/pull/246, for now I'd suggest building package sequentially, easiest way is to run `./src/depthai-ros/build.sh` in your workspace directory, this executes colcon with some additional arguments for optimization, you can check out contents [here](https://github.com/luxonis/depthai-ros/blob/humble/build.sh) if you want to alter it

qnaa42, 2023-03-16:
Thank you! That allows me to move forward. The command that fixed it for me was:

rm -rf /home/mateusz/dai_ws/build && ./src/depthai-ros/build.sh

Serafadam, 2023-03-17:
Happy to help!

closed 2023-03-17 by Serafadam