luxonis/depthai-ros #252: roslaunch depthai_examples stereo_inertial_node.launch error

TDC-L, 2023-03-16:
my platform is: Xavier AGX + OAK-D-Pro-Poe
When I launch this
roslaunch depthai_examples stereo_inertial_node.launch

I have get the following error

process[rviz-7]: started with pid [154660]
[ INFO] [1678972948.937030863]: Initializing nodelet with 6 worker threads.
 nnPath ,, /home/ubuntu/dai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
1280 720 1280 720
Listing available devices...
Device Mx ID: 19443010E1FF721300
[19443010E1FF721300] [169.254.1.222] [6.390] [SpatialDetectionNetwork(9)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 154646, exit code -11, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/9716874c-c3fd-11ed-a061-48b02d5112cc/stereo_inertial_publisher-3.log].
log file: /home/ubuntu/.ros/log/9716874c-c3fd-11ed-a061-48b02d5112cc/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================

Serafadam, 2023-03-16:
Hi, can you run [demo script](https://docs.luxonis.com/en/latest/#demo-script) successfully? If yes, please run `export DEPTHAI_DEBUG=1` before running launch file again to get some more logs.

TDC-L, 2023-03-17:
Yeah,of course run demo script successfully.

I try to run export DEPTHAI_DEBUG=1 before running launch file again and get 
these logs:
[ INFO] [1679020220.232178191]: Initializing nodelet with 6 worker threads.
 nnPath ,, /home/ubuntu/dai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
[2023-03-17 10:30:20.615] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-03-17 10:30:20.615] [debug] Library information - version: 2.20.2, commit:  from , build: 2023-02-02 20:08:23 +0000
[2023-03-17 10:30:20.624] [debug] Initialize - finished
1280 720 1280 720
[2023-03-17 10:30:20.832] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 5ms, archive read: 210ms
Listing available devices...
Device Mx ID: 19443010E1FF721300
[2023-03-17 10:30:21.154] [debug] Device - OpenVINO version: 2022.1
[2023-03-17 10:30:21.155] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 10 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 00 bb 00
[2023-03-17 10:30:21.602] [debug] Resources - Archive 'depthai-device-fwp-8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd.tar.xz' open: 6ms, archive read: 979ms
[2023-03-17 10:30:21.625] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=19443010E1FF721300, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-03-17 10:30:21.689] [debug] Connected bootloader version 0.0.19
[2023-03-17 10:30:21.689] [info] New bootloader version available. Device has: 0.0.19, available: 0.0.24
[2023-03-17 10:30:22.583] [debug] Booting FW with Bootloader. Version 0.0.19, Time taken: 893ms
[2023-03-17 10:30:22.583] [debug] DeviceBootloader about to be closed...
[2023-03-17 10:30:22.583] [debug] XLinkResetRemote of linkId: (0)
[2023-03-17 10:30:23.831] [debug] DeviceBootloader closed, 1248
[2023-03-17 10:30:23.835] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=19443010E1FF721300, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[19443010E1FF721300] [169.254.1.222] [4.840] [system] [warning] PRINT:LeonCss: GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 7
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
DepthAI Firmware - version: 8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Found a device/board entry matching the eeprom data - Board: OAK-D S2 PoE (generic) Device: 
Board options: 00000007
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: c2 25 3a
SR CR: 40 07
QUAD mode already enabled. Dummy cfg: 0x00
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Setup network with bootloader config
OsDrvCprSysDeviceAction() for PCIe failed: 13
nexus0: <RTEMS Nexus device>
pcib0: <MyriadX PCIe RC> on nexus0
PCIE_REGS_PCIE_CFG_ADR = 0x60100
[... 302 lines cut ...]
Starting Guzzi command handling loop...
inc_camera_process set exposure and gain
[19443010E1FF721300] [169.254.1.222] [6.467] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
finished sippPrePrepareLinesFreeRT
[19443010E1FF721300] [169.254.1.222] [6.478] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
finished sippPrePrepareLinesFreeRT
[19443010E1FF721300] [169.254.1.222] [6.511] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[19443010E1FF721300] [169.254.1.222] [6.577] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
[19443010E1FF721300] [169.254.1.222] [6.788] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 5429, exit code -11, cmd /home/ubuntu/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/ubuntu/.ros/log/a6de9f2a-c46b-11ed-8129-48b02d5112cc/stereo_inertial_publisher-3.log].
log file: /home/ubuntu/.ros/log/a6de9f2a-c46b-11ed-8129-48b02d5112cc/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[oak_state_publisher-2] killing on exit
[stereo_inertial_publisher-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

TDC-L, 2023-03-17:
Thanks for your help.

1.Yeah,of course is the latest branch.
2.I can also run roslaunch depthai_ros_driver camera.launch .

Serafadam, 2023-03-22:
Hi, quick update, I'm doing some tests with PoE devices in general to find out this and other issues that can happen in different configurations, though if `depthai_ros_driver` works for you then I suggest using this as the default, since it has more features and is more actively maintained. Is there a particular reason that you've chosen `stereo_inertial_node` over the other driver?