luxonis/depthai-ros #257: [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at

MAVProxyUser, 2023-03-18:
Unable to stream from camera, Black screen. 

```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-18-12-33-05-331218-ubuntu-539149
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [539163]
[INFO] [cameracalibrator-2]: process started with pid [539165]
[component_container-1] [INFO] [1679157186.574037766] [oak_container]: Load Library: /home/ubuntu/luxonis_oak_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1679157186.754660280] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1679157186.754853081] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [2023-03-18 12:33:06.787] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[component_container-1] [2023-03-18 12:33:06.789] [debug] Library information - version: 2.20.2, commit: 5ead66c017593563c8f730847c3e7ed2df8e950c from 2023-03-08 15:39:34 +0200, build: 2023-03-18 06:24:30 +0000
[component_container-1] [2023-03-18 12:33:06.795] [debug] Initialize - finished
[component_container-1] [INFO] [1679157186.795753948] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [2023-03-18 12:33:07.000] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 10ms, archive read: 197ms
[component_container-1] [2023-03-18 12:33:07.014] [debug] Device - OpenVINO version: universal
[component_container-1] [2023-03-18 12:33:07.018] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
[component_container-1] libnop:
[component_container-1] 0000: b9 10 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
[component_container-1] 0020: be be be 01 00 bb 00
[component_container-1] [2023-03-18 12:33:07.577] [debug] Resources - Archive 'depthai-device-fwp-8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd.tar.xz' open: 7ms, archive read: 778ms
[component_container-1] [2023-03-18 12:33:08.323] [debug] Searching for booted device: DeviceInfo(name=1.2.1, mxid=1844301091E6331300, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-1] [INFO] [1679157189.311138631] [oak]: Camera with MXID: 1844301091E6331300 and Name: 1.2.1 connected!
[component_container-1] [1844301091E6331300] [1.2.1] [0.956] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0xFCCBD2D0
[component_container-1] GPIO boot mode 0x3, interface SPI_MASTER_EFF
[component_container-1] Setting aons(0..4) back to boot from flash (offset = 0)
[component_container-1] ====ENABLE WATCHDOG====1
[component_container-1] initial keepalive, countdown: 10
[component_container-1] PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
[component_container-1] Board init ret 3
[component_container-1] DepthAI Firmware - version: 8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd
[component_container-1] eeprom configuration version: 55AA0007
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] eeprom configuration load from user area, status: 0
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] PMIC read  reg 0x19, val 0xb8
[component_container-1] PMIC read  reg 0x23, val 0x10
[component_container-1] PMIC read  reg 0x31, val 0x08
[component_container-1] PMIC read  reg 0x33, val 0x28
[component_container-1] Found a device/board entry matching the eeprom data - Board: OAK-D S2/Pro (R6 - generic) Device: 
[component_container-1] Board options: 00000004
[component_container-1] --> brdInit ...
[component_container-1] brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]
[component_container-1] 
[component_container-1] spi_N25Q_init: Flash JEDEC ID: c2 25 3a
[component_container-1] SR CR: 40 07
[component_container-1] QUAD mode already enabled. Dummy cfg: 0x00
[component_container-1] Initialized pin: 56, mode: 7, level: 1, pad: 0x00000010
[component_container-1] Opening bus for IR driver: 1
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Opening bus for IR driver: 2
[component_container-1] LM3644 detected, ID = 0x02
[component_container-1] ===== IR write bus 2: 0x07 = 0x89
[component_container-1] ===== IR write bus 2: 0x02 = 0x01
[component_container-1] ===== IR write bus 2: 0x03 = 0x00
[component_container-1] ===== IR write bus 2: 0x04 = 0x00
[component_container-1] ===== IR write bus 2: 0x05 = 0x00
[component_container-1] ===== IR write bus 2: 0x06 = 0x00
[component_container-1] ===== IR write bus 2: 0x07 = 0x09
[component_container-1] ===== IR write bus 2: 0x08 = 0x1a
[component_container-1] ===== IR write bus 2: 0x09 = 0x08
[component_container-1] ===== IR write bus 2: 0x01 = 0x24
[component_container-1] Opening bus for IR driver: 3
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Closing EEPROm!
[component_container-1] MyriaX board configuration
[component_container-1] pll0 frequency: 700000, ref0 frequency: 24000
[component_container-1] Is booted from flash by bootloader: 0
[component_container-1] Networking not available...
[component_container-1] === Enumerating on socket: Cam_A / RGB / Center
[... 31 lines cut ...]
[component_container-1] Usb connection speed: Super - USB 3.0
[component_container-1] Temperature: Driver registered.
[component_container-1] Temperature: Initialized driver.
[component_container-1] Temperature: Sensor opened: CSS.
[component_container-1] Temperature: Sensor opened: MSS.
[component_container-1] Temperature: Sensor opened: UPA.
[component_container-1] Temperature: Sensor opened:
[component_container-1] [INFO] [1679157189.313438512] [oak]: USB SPEED: SUPER
[component_container-1] [1844301091E6331300] [1.2.1] [0.966] [system] [info] Memory Usage - DDR: 0.12 / 340.42 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 7.00 / 77.32 MiB, LeonRT Heap: 2.89 / 41.23 MiB
[component_container-1] [1844301091E6331300] [1.2.1] [0.966] [system] [info] Temperatures - Average: 36.53 °C, CSS: 37.48 °C, MSS 36.30 °C, UPA: 36.06 °C, DSS: 36.30 °C
[component_container-1] [1844301091E6331300] [1.2.1] [0.966] [system] [info] Cpu Usage - LeonOS 65.40%, LeonRT: 1.69%
[component_container-1] [1844301091E6331300] [1.2.1] [0.967] [system] [warning] PRINT:LeonCss:  DSS.
[component_container-1] [1844301091E6331300] [1.2.1] [0.978] [system] [warning] PRINT:LeonCss: I: [Timesync] [   4202637] [XLin] startSync:130     Timesync | Callback not set
[component_container-1] [1844301091E6331300] [1.2.1] [0.989] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] [INFO] [1679157189.347519480] [oak]: Device type: OAK-D-PRO-AF
[component_container-1] [INFO] [1679157189.354222803] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-18 12:33:09.355] [debug] Device about to be closed...
[component_container-1] [2023-03-18 12:33:09.520] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-18 12:33:09.522] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-18 12:33:09.522] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-18 12:33:10.789] [debug] Device closed, 1434
[component_container-1] [ERROR] [1679157190.815125257] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

MAVProxyUser, 2023-03-18:
```ubuntu@ubuntu:~$ cat ~/.bashrc | tail 
    . /etc/bash_completion
  fi
fi
export PATH=/usr/local/cuda-11.4/bin:$PATH
export LD_LIBRARY_PATH=/usr/local/cuda-11.4/lib64:$LD_LIBRARY_PATH
source /home/ubuntu/ros2_humble/install/setup.sh
source /home/ubuntu/luxonis_oak_ws/install/setup.sh
source /home/ubuntu/jetson_stats_ws/install/setup.sh
export OPENBLAS_CORETYPE=ARMV8
export DEPTHAI_DEBUG=1

ubuntu@ubuntu:~$ cat /etc/issue
Ubuntu 20.04.6 LTS \n \l

ubuntu@ubuntu:~$ uname -a 
Linux ubuntu 5.10.104-tegra #1 SMP PREEMPT Tue Jan 24 15:09:44 PST 2023 aarch64 aarch64 aarch64 GNU/Linux
```

MAVProxyUser, 2023-03-18:
```
ubuntu@ubuntu:~$ cd depthai/
ubuntu@ubuntu:~/depthai$ git log | head
commit be73fbc4c8dd355366edf7bc1782387b421eeb83
Merge: 32879ee5 22192e78
Author: Pastukhov Daniil <plus79222222238@gmail.com>
Date:   Fri Feb 10 12:50:03 2023 +0100

    Merge pull request #938 from luxonis/fix/cv2_cond_import
    
    Fix cv2 import

commit 22192e786bf0a11b79aa08f932f259d39ebf690e
ubuntu@ubuntu:~/depthai$ git branch
* main
ubuntu@ubuntu:~/depthai$ cd ../luxonis_oak_ws/src/depthai-ros/
ubuntu@ubuntu:~/luxonis_oak_ws/src/depthai-ros$ git log | head 
commit 29cf4a0140d211b7b871c93198f3433b648933ae
Author: Adam Serafin <adam.serafin@luxonis.com>
Date:   Fri Mar 17 11:11:29 2023 +0100

    Build time improvements for Humble (#246)
    
    * code cleanup to improve build times

commit 49152be9afcbfdb5749a43ff9531a8a793c55917
Author: Serafadam <serafadam@gmail.com>
ubuntu@ubuntu:~/luxonis_oak_ws/src/depthai-ros$ git branch
* humble
```

MAVProxyUser, 2023-03-18:
This seems to be a GUI issue primarily. The rest of the ROS backend seems fine. I'm able to pull YOLO hits off the example. 
[image]

```
ubuntu@ubuntu:~$ ros2 launch depthai_examples yolov4_publisher.launch.py
...
ubuntu@ubuntu:~$ ros2 topic list
/color/camera_info
/color/image
/color/yolov4_Spatial_detections
/joint_states
/parameter_events
/robot_description
/rosout
/stereo/camera_info
/stereo/depth
/tf
/tf_static
```

MAVProxyUser, 2023-03-20:
@Serafadam 
```
$  ros2 launch depthai_ros_driver  camera.launch.py
```

gives the following 
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-20-10-20-12-813752-ubuntu-12299
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [12314]
[INFO] [component_container-2]: process started with pid [12316]
[component_container-2] [INFO] [1679322015.906563135] [oak_container]: Load Library: /home/ubuntu/luxonis_oak_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[robot_state_publisher-1] [INFO] [1679322016.033384462] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1679322016.033751696] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1679322016.033823920] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1679322016.033880433] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1679322016.033963793] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1679322016.034020657] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1679322016.034108210] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1679322016.034168466] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1679322016.034221395] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1679322016.034273043] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1679322016.104666068] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1679322016.104877941] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [2023-03-20 10:20:16.148] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[component_container-2] [2023-03-20 10:20:16.150] [debug] Library information - version: 2.20.2, commit: 5ead66c017593563c8f730847c3e7ed2df8e950c from 2023-03-08 15:39:34 +0200, build: 2023-03-18 06:24:30 +0000
[component_container-2] [2023-03-20 10:20:16.158] [debug] Initialize - finished
[component_container-2] [INFO] [1679322016.158837300] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [2023-03-20 10:20:16.326] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 7ms, archive read: 164ms
[component_container-2] [2023-03-20 10:20:16.383] [debug] Device - OpenVINO version: universal
[component_container-2] [2023-03-20 10:20:16.389] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
[component_container-2] libnop:
[component_container-2] 0000: b9 10 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
[component_container-2] 0020: be be be 01 00 bb 00
[component_container-2] [2023-03-20 10:20:16.926] [debug] Resources - Archive 'depthai-device-fwp-8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd.tar.xz' open: 8ms, archive read: 765ms
[component_container-2] [2023-03-20 10:20:17.676] [debug] Searching for booted device: DeviceInfo(name=1.2.4, mxid=1844301011A9460E00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-2] [1844301011A9460E00] [1.2.4] [0.966] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0xF8135BD0
[component_container-2] GPIO boot mode 0x16, interface USBD
[component_container-2] Setting aons(0..4) back to boot from flash (offset = 0)
[component_container-2] ====ENABLE WATCHDOG====1
[component_container-2] initial keepalive, countdown: 10
[component_container-2] PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
[component_container-2] Board init ret 3
[component_container-2] DepthAI Firmware - version: 8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd
[component_container-2] eeprom configuration version: 55AA0007
[component_container-2] Reading VERSION 7 --- ->
[component_container-2] eeprom configuration load from user area, status: 0
[component_container-2] Reading VERSION 7 --- ->
[component_container-2] Didn't find any boards matching the eeprom data
[component_container-2] Board options: 00000000
[component_container-2] --> brdInit ...
[component_container-2] brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]
[component_container-2] 
[component_container-2] spi_N25Q_init: Flash JEDEC ID: ff ff ff
[component_container-2] Invalid Flash JEDEC ID... No NOR available
[component_container-2] Could not register the spi device
[component_container-2] Opening bus for IR driver: 1
[component_container-2] Failed to probe IR driver LM3644
[component_container-2] Opening bus for IR driver: 2
[component_container-2] Failed to probe IR driver LM3644
[component_container-2] Opening bus for IR driver: 3
[... cut ...]
[component_container-2] [2023-03-20 10:20:19.711] [debug] XLinkResetRemote of linkId: (0)
[component_container-2] [2023-03-20 10:20:20.152] [debug] Device closed, 1443
[component_container-2] [ERROR] [1679322020.174698398] [oak_container]: Component constructor threw an exception: Sensor not supported!
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: Sensor not supported!
```

Serafadam, 2023-03-22:
Thanks for the logs, it seems that device returns older sensor name, I added a fix in https://github.com/luxonis/depthai-ros/pull/247, please try `depthai_filters_humble` branch and let me know if that fix works for you

MAVProxyUser, 2023-03-22:
I needed  libopencv-dev & libopencv-contrib-dev, but my version of ubuntu had a known issue. Solution was:
```
sudo apt install  libopencv-calib3d-dev libopencv-contrib-dev libopencv-features2d-dev libopencv-highgui-dev libopencv-imgcodecs-dev  libopencv-objdetect-dev  libopencv-shape-dev libopencv-stitching-dev  libopencv-superres-dev  libopencv-video-dev libopencv-videoio-dev  libopencv-videostab-dev  libopencv4.2-java libopencv-calib3d4.2 libopencv-contrib4.2 libopencv-features2d4.2 libopencv-highgui4.2 libopencv-imgcodecs4.2 libopencv-videoio4.2 libgdal26 libodbc1
```
per: https://github.com/appveyor/ci/issues/3434

MAVProxyUser, 2023-03-22:
I had to install several packages as part of a workaround to a known problem on Ubuntu 20.04 as described here:
https://github.com/appveyor/ci/issues/3434

First I had to 
```
sudo apt install  libopencv-calib3d-dev libopencv-contrib-dev libopencv-features2d-dev libopencv-highgui-dev libopencv-imgcodecs-dev  libopencv-objdetect-dev  libopencv-shape-dev libopencv-stitching-dev  libopencv-superres-dev  libopencv-video-dev libopencv-videoio-dev  libopencv-videostab-dev  libopencv4.2-java libopencv-calib3d4.2 libopencv-contrib4.2 libopencv-features2d4.2 libopencv-highgui4.2 libopencv-imgcodecs4.2 libopencv-videoio4.2 libgdal26 libodbc1
```
then I was able to 
```
sudo apt-get install libopencv-dev  
sudo apt-get install  libopencv-contrib-dev
```

Now I'm dealing with OpenCV issues. 

```
--- stderr: image_geometry                                                                                                                    
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

```

Will check in when I solve that.

MAVProxyUser, 2023-03-23:
ok @Serafadam ... On my Oak-1-MAX that seemed to sort tings out
```
[component_container-2] [E] app_guzzi_command_callback():218: command "5 0" sent
[component_container-2] 
[component_container-2] Starting Guzzi command handling loop...
[component_container-2] [1844301011A9460E00] [1.2.1] [1.197] [system] [warning] PRINT:LeonCss: inc_camera_process set exposure and gain
[component_container-2] [INFO] [1679540113.897492909] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [1844301011A9460E00] [1.2.1] [1.252] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301011A9460E00] [1.2.1] [1.256] [XLinkOut(1)] [info] Increasing 'rgb_isp' stream size from 5242880B to 5580800B
[component_container-2] [1844301011A9460E00] [1.2.1] [1.538] [system] [warning] PRINT:LeonCss: initial keepalive, countdown: 8
[component_container-2] [1844301011A9460E00] [1.2.1] [1.659] [IMU(3)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
[component_container-2] [2023-03-22 22:55:14.840] [debug] DataOutputQueue (imu_imu) closed
[component_container-2] [2023-03-22 22:55:14.840] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-2] [2023-03-22 22:55:14.840] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-2] [2023-03-22 22:55:14.841] [debug] DataOutputQueue (rgb_isp) closed
[component_container-2] [2023-03-22 22:55:15.160] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
```

MAVProxyUser, 2023-03-23:
Well looking at my S2... I guess I'm wrong, because there I keep seeing the stream of Men / Temp /Cpu usage..
```
[component_container-2] Starting Guzzi command handling loop...
[component_container-2] inc_camera_process set exposure and gain
[component_container-2] [1944301041A31F1300] [1.2.1] [4.519] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [1944301041A31F1300] [1.2.1] [4.530] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] [1944301041A31F1300] [1.2.1] [4.541] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1944301041A31F1300] [1.2.1] [4.552] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1944301041A31F1300] [1.2.1] [4.563] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [INFO] [1679540325.342977923] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [1944301041A31F1300] [1.2.1] [4.969] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.41 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
[component_container-2] [1944301041A31F1300] [1.2.1] [4.969] [system] [info] Temperatures - Average: 31.53 °C, CSS: 32.49 °C, MSS 31.29 °C, UPA: 31.53 °C, DSS: 30.80 °C
[component_container-2] [1944301041A31F1300] [1.2.1] [4.969] [system] [info] Cpu Usage - LeonOS 52.53%, LeonRT: 42.24%
[component_container-2] [1944301041A31F1300] [1.2.1] [5.971] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.41 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
[component_container-2] [1944301041A31F1300] [1.2.1] [5.971] [system] [info] Temperatures - Average: 32.43 °C, CSS: 33.69 °C, MSS 32.49 °C, UPA: 32.01 °C, DSS: 31.53 °C
[component_container-2] [1944301041A31F1300] [1.2.1] [5.971] [system] [info] Cpu Usage - LeonOS 79.94%, LeonRT: 52.96%
[component_container-2] [1944301041A31F1300] [1.2.1] [6.973] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.41 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
[component_container-2] [1944301041A31F1300] [1.2.1] [6.973] [system] [info] Temperatures - Average: 33.27 °C, CSS: 33.93 °C, MSS 32.97 °C, UPA: 33.93 °C, DSS: 32.25 °C
[component_container-2] [1944301041A31F1300] [1.2.1] [6.973] [system] [info] Cpu Usage - LeonOS 77.88%, LeonRT: 52.68%
[component_container-2] [1944301041A31F1300] [1.2.1] [7.975] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.41 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
[component_container-2] [1944301041A31F1300] [1.2.1] [7.975] [system] [info] Temperatures - Average: 33.68 °C, CSS: 34.64 °C, MSS 33.21 °C, UPA: 34.40 °C, DSS: 32.49 °C
[component_container-2] [1944301041A31F1300] [1.2.1] [7.975] [system] [info] Cpu Usage - LeonOS 75.56%, LeonRT: 52.37%
[component_container-2] [1944301041A31F1300] [1.2.1] [8.976] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.41 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
```

But on my Pro, I see:
```
[component_container-2] Starting Guzzi command handling loop...
[component_container-2] inc_camera_process set exposure and gain
[component_container-2] [1844301091E6331300] [1.2.1] [4.855] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [1844301091E6331300] [1.2.1] [4.866] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301091E6331300] [1.2.1] [4.877] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301091E6331300] [1.2.1] [4.899] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [1844301091E6331300] [1.2.1] [4.899] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301091E6331300] [1.2.1] [4.964] [system] [info] Memory Usage - DDR: 201.40 / 340.42 MiB, CMX: 2.47 / 2.50 MiB, LeonOS Heap: 49.15 / 77.32 MiB, LeonRT Heap: 6.29 / 41.23 MiB
[component_container-2] [1844301091E6331300] [1.2.1] [4.964] [system] [info] Temperatures - Average: 33.09 °C, CSS: 33.93 °C, MSS 32.01 °C, UPA: 33.45 °C, DSS: 32.97 °C
[component_container-2] [1844301091E6331300] [1.2.1] [4.964] [system] [info] Cpu Usage - LeonOS 31.50%, LeonRT: 28.91%
[component_container-2] [2023-03-22 23:00:49.161] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-2] [2023-03-22 23:00:49.162] [debug] DataOutputQueue (nn_nn) closed
[component_container-2] [2023-03-22 23:00:49.167] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-2] [2023-03-22 23:00:49.170] [debug] DataOutputQueue (stereo_stereo) closed
[component_container-2] [2023-03-22 23:00:49.176] [debug] DataOutputQueue (rgb_isp) closed
[component_container-2] [2023-03-22 23:00:49.176] [debug] Device about to be closed...
[... cut ...]
[component_container-2] [2023-03-22 23:01:33.420] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-2] [2023-03-22 23:01:33.421] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-2] [2023-03-22 23:01:33.720] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
```

So things are closer I guess, but not fully right.

MAVProxyUser, 2023-03-23:
Here we go @Serafadam  I'm noticing now that this node needs configured via ./install/depthai_ros_driver/share/depthai_ros_driver/config/camera.yaml   

setting the following made the OAK-1 cameras happy. 
```
/oak:
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: false
```

The pro only works if I set: 
```
      i_enable_ir: false
```

Else if the IR setting is enabled it fails saying: 
```
[component_container-2] Starting Guzzi command handling loop...
[component_container-2] [1844301091E6331300] [1.2.1] [5.325] [system] [warning] PRINT:LeonCss: inc_camera_process set exposure and gain
[component_container-2] [1844301091E6331300] [1.2.1] [5.347] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [1844301091E6331300] [1.2.1] [5.358] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301091E6331300] [1.2.1] [5.369] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] [1844301091E6331300] [1.2.1] [5.380] [system] [warning] PRINT:LeonCss: Reading..........Reading VERSION 7 --- ->
[component_container-2] [1844301091E6331300] [1.2.1] [5.391] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [2023-03-23 01:15:49.621] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-2] [2023-03-23 01:15:49.621] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-2] [2023-03-23 01:15:49.621] [debug] DataOutputQueue (nn_nn) closed
[component_container-2] [2023-03-23 01:15:49.630] [debug] DataOutputQueue (stereo_stereo) closed
[component_container-2] [2023-03-23 01:15:49.630] [debug] Device about to be closed...
[component_container-2] [2023-03-23 01:15:49.630] [debug] DataOutputQueue (rgb_isp) closed
[component_container-2] [2023-03-23 01:15:49.630] [debug] XLinkResetRemoteTimeout returned: X_LINK_ERROR
[component_container-2] [2023-03-23 01:15:50.295] [debug] XLinkResetRemote of linkId: (0)
[component_container-2] [2023-03-23 01:15:51.052] [debug] Device closed, 1422
[component_container-2] [2023-03-23 01:15:51.053] [debug] DataOutputQueue (imu_imu) closed
[component_container-2] [2023-03-23 01:15:51.053] [debug] DataInputQueue (right_control) closed
[component_container-2] [2023-03-23 01:15:51.054] [debug] DataInputQueue (left_control) closed
[component_container-2] [2023-03-23 01:15:51.054] [debug] DataInputQueue (rgb_control) closed

```

MAVProxyUser, 2023-03-23:
Also FWIW before you had me run the camera.launch.py I had been trying 
```
ros2 launch depthai_ros_driver calibration.launch.py camera_model:=OAK-D-PRO
```

And I still get:
```
...
[component_container-1] Initializing XLink...
[component_container-1] UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
[component_container-1] UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
[component_container-1] UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
[component_container-1] initial keepalive, countdown: 9
[component_container-1] UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
[component_container-1] Done!
[component_container-1] Usb connection speed: Super - USB 3.0
[component_container-1] [1844301091E6331300] [1.2.1] [0.961] [system] [warning] PRINT:LeonCss: Temperature: Driver registered.
[component_container-1] Temperature: Initialized driver.
[component_container-1] Temperature: Sensor opened: CSS.
[component_container-1] Temperature: Sensor opened: MSS.
[component_container-1] Temperature: Sensor opened: UPA.
[component_container-1] Temperature: Sensor opened: DSS.
[component_container-1] [1844301091E6331300] [1.2.1] [0.962] [system] [info] Memory Usage - DDR: 0.12 / 340.42 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 7.25 / 77.32 MiB, LeonRT Heap: 2.89 / 41.23 MiB
[component_container-1] [1844301091E6331300] [1.2.1] [0.962] [system] [info] Temperatures - Average: 37.06 °C, CSS: 37.71 °C, MSS 37.01 °C, UPA: 36.77 °C, DSS: 36.77 °C
[component_container-1] [1844301091E6331300] [1.2.1] [0.963] [system] [info] Cpu Usage - LeonOS 71.07%, LeonRT: 1.69%
[component_container-1] [1844301091E6331300] [1.2.1] [0.983] [system] [warning] PRINT:LeonCss: I: [Timesync] [   5207311] [XLin] startSync:130     Timesync | Callback not set
[component_container-1] [INFO] [1679549259.226435936] [oak]: Camera with MXID: 1844301091E6331300 and Name: 1.2.1 connected!
[component_container-1] [INFO] [1679549259.228449989] [oak]: USB SPEED: SUPER
[component_container-1] [1844301091E6331300] [1.2.1] [1.005] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] [INFO] [1679549259.258616421] [oak]: Device type: OAK-D-PRO-AF
[component_container-1] [INFO] [1679549259.264823795] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-23 01:27:39.265] [debug] Device about to be closed...
[component_container-1] [2023-03-23 01:27:39.423] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:27:39.423] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:27:39.936] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:27:40.083] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-23 01:27:40.682] [debug] Device closed, 1416
[component_container-1] [ERROR] [1679549260.701787006] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

MAVProxyUser, 2023-03-23:
calibration.launch.py also still fails with my S2
```
[component_container-1] [1944301041A31F1300] [1.2.1] [1.002] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] [INFO] [1679549371.902162186] [oak]: Device type: OAK-D-S2-AF
[component_container-1] [INFO] [1679549371.908023069] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-23 01:29:31.908] [debug] Device about to be closed...
[component_container-1] [2023-03-23 01:29:32.060] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:29:32.061] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:29:32.062] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-23 01:29:33.326] [debug] Device closed, 1417
[component_container-1] [ERROR] [1679549373.346861649] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

MAVProxyUser, 2023-03-23:
original Oak-1 fails... (note the  "Device type" type here) @Serafadam 
```
[component_container-1] [INFO] [1679549436.793553763] [oak]: Camera with MXID: 14442C10A18652D000 and Name: 1.2.1 connected!
[component_container-1] [INFO] [1679549436.796419918] [oak]: USB SPEED: SUPER
[component_container-1] [14442C10A18652D000] [1.2.1] [0.983] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] [INFO] [1679549436.807773594] [oak]: Device type: 
[component_container-1] [INFO] [1679549436.813356047] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-23 01:30:36.814] [debug] Device about to be closed...
[component_container-1] [2023-03-23 01:30:36.972] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:30:36.972] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:30:37.510] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:30:37.831] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-23 01:30:38.250] [debug] Device closed, 1435
[component_container-1] [ERROR] [1679549438.274026895] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

MAVProxyUser, 2023-03-23:
You can see the OAK-1-MAX does report a name under "Device type"
```
[component_container-1] [INFO] [1679549569.647442958] [oak]: Camera with MXID: 1844301011A9460E00 and Name: 1.2.1 connected!
[component_container-1] [INFO] [1679549569.649557688] [oak]: USB SPEED: SUPER
[component_container-1] [1844301011A9460E00] [1.2.1] [1.015] [system] [warning] PRINT:LeonCss: Reading from Factory EEPROM contents
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] [INFO] [1679549569.669528759] [oak]: Device type: OAK-1-MAX
[component_container-1] [INFO] [1679549569.677008122] [oak]: Pipeline type: RGBStereo
[component_container-1] [2023-03-23 01:32:49.678] [debug] Device about to be closed...
[component_container-1] [2023-03-23 01:32:49.679] [debug] XLinkResetRemote of linkId: (0)
[component_container-1] [2023-03-23 01:32:49.680] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:32:49.680] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[component_container-1] [2023-03-23 01:32:51.110] [debug] Device closed, 1432
[component_container-1] [ERROR] [1679549571.128903967] [oak_container]: Component constructor threw an exception: _Map_base::at
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: _Map_base::at
```

larshg, 2023-03-23:
Hey @MAVProxyUser and @Serafadam 

Today I tried to launched the calibration with: depthai-ros/depthai_ros_driver/launch/calibration.launch

And I see similar errors.

ROS: Noetic
Depthai repo on noetic branch.

When I run it with debug I get the following callstack and a segmentation fault:
[image]

I tried to disable camera_i_enable_ir, but I couldn't disable it by modifying any config files, neither config/camera.yaml nor cfg/paramemers.cfg seemed to do the trick, so I ended up modyfying the header file:
[image]

Maybe this can give some hints to where we can look/change?

larshg, 2023-03-23:
Running roslaunch depthai_examples stereo_inertial_node.launch works fine, so the device is functional :smile:

larshg, 2023-03-23:
I guess mine is more similar to #245.

Except I run on laptop in docker.

Serafadam, 2023-03-23:
@MAVProxyUser @larshg 
Ok, I think I've pinpointed that issue, and of course it's a typo :smiling_face_with_tear: :facepalm: 
Until fix is added, easiest way to get rid of the issue is to edit calibration launch file, change pipeline_type parameter from "RGBStereo" to "RGBDStereo".

larshg, 2023-03-23:
> 

Indeed it launches now without crashing, but I don't get any images still. There doesn't seem to be anything published to /oak/rgb/image_raw.

Serafadam, 2023-03-23:
Is this also happening if you run other launches in depthai_ros_driver?

MAVProxyUser, 2023-03-23:
@larshg I too was able to run the Stereo Internal Node launch file just fine. 
```
$  ros2 launch depthai_examples  stereo_inertial_node.launch.py camera_model:=OAK-D-PRO
```

MAVProxyUser, 2023-03-23:
Thanks @Serafadam I'll try the suggested fix. 
```
diff --git a/depthai_ros_driver/launch/calibration.launch.py b/depthai_ros_driver/launch/calibration.launch.py
index 98f682a..b244b8a 100644
--- a/depthai_ros_driver/launch/calibration.launch.py
+++ b/depthai_ros_driver/launch/calibration.launch.py
@@ -48,7 +48,7 @@ def launch_setup(context, *args, **kwargs):
                         package="depthai_ros_driver",
                         plugin="depthai_ros_driver::Camera",
                         name=name,
-                        parameters=[{"camera.i_pipeline_type": "RGBStereo",
+                        parameters=[{"camera.i_pipeline_type": "RGBDStereo",
                                      "camera.i_nn_type": "none",
                                      "right.i_publish_topic": True,
                                      "left.i_publish_topic": True}],
```
Making sure it was in the install file of course 
```
ubuntu@ubuntu:~$ grep RGBDStereo . -r 
./luxonis_oak_ws/install/depthai_ros_driver/share/depthai_ros_driver/launch/calibration.launch.py:                        parameters=[{"camera.i_pipeline_type": "RGBDStereo",
./luxonis_oak_ws/src/depthai-ros/depthai_ros_driver/launch/calibration.launch.py:                        parameters=[{"camera.i_pipeline_type": "RGBDStereo",
```

I am now getting the cam streaming, but still getting an error

[image]

larshg, 2023-03-24:
> Is this also happening if you run other launches in depthai_ros_driver?

I tried running some other example which seemed to work (I ran the RGB-D-pcl.launch, which published on the /oak/points topic with 30 hz). Then I reran the calibration and now it shows an grayscale image of the center camera.
Now I just need to figure how to do the actual calibration :smile: 

Not entirely sure why it shows an image now :confused: 

Sidenote: I initially connected two USB-C with the supplied splitter, to power the IR-emitter, to see if that had any changes. But now I returned to just a single USB-C and it still works.

MAVProxyUser, 2023-03-24:
```
Now I just need to figure how to do the actual calibration 😄
```
*same*!

larshg, 2023-03-24:
> ```
> Now I just need to figure how to do the actual calibration 😄
> ```
> 
> _same_!

So I set up the size paramemter to be 8x5, which match the number of intersections on the calibration plate and also set the square size to 4.0 (cm) and now it correctly detects the calibration board.

Waiting for the result after pressing calibrate :smile:

larshg, 2023-03-24:
@Serafadam I noticed that the calibration is done images with resolution of 1280x720 - is there a way to make use the full resolution?

larshg, 2023-03-24:
@Serafadam I assume mono is the RGB camera - the center one? Alternative to mono, is stereo right -  which is not what I want to do.

I'll try the `i_set_isp_scale`

Serafadam, 2023-03-24:
You can change resolution with `i_resolution` parameter

MAVProxyUser, 2023-03-25:
@Serafadam I put the Oak-1 bug here, since I figured out what it is related to
https://github.com/luxonis/depthai-core/issues/782
Pretty well these lines of code failing due to the blank camera name returned.  
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L295
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L308
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L329
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L344
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L351
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L373
https://github.com/luxonis/depthai-core/blob/5ead66c017593563c8f730847c3e7ed2df8e950c/src/device/CalibrationHandler.cpp#L418

jaggiK, 2023-05-01:
Hi,
I am facing the same issue with OAK-D Lite. Is this issue resolved? Thanks

Serafadam, 2023-09-14:
Closing the issue as it seems problems were resolved, please reopen if you have further comments/questions.

closed 2023-09-14 by Serafadam