luxonis/depthai-ros #258: [BUG] ROS message Detection3DArray has incomplete bbox data

RemiFabre, 2023-03-19:
**Describe the bug**
While doing spatial Yolo, I noticed the y size of the bbox is not updated in the ROS message:
`  bbox:
    center:
      position:
        x: 136.5
        y: 222.5
        z: 0.0
      orientation:
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    size:
      x: 243.0
      y: 0.0
      z: 0.0
`

**Minimal Reproducible Example**
Start a spatial yolo and:
`ros2 topic echo /oak/nn/detections`

**Fix**
It seems to be a simple typo, I fixed it with the following changes.

In the file `depthai_bridge/src/SpatialDetectionConverter.cpp`, 
in the function `toRosMsg`, add:
`opDetectionMsg.detections[i].bbox.size_y = ySize;`

in the function `toRosVisionMsg`, add:
`opDetectionMsg.detections[i].bbox.size.y = ySize;`

Serafadam, 2023-03-20:
Hi, thanks, this will be added as part of https://github.com/luxonis/depthai-ros/pull/247

Serafadam, 2023-03-27:
Added in https://github.com/luxonis/depthai-ros/pull/247

closed 2023-03-27 by Serafadam