luxonis/depthai-ros #263: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'

Embedded1993, 2023-03-28:
OS: butuntu 20.04 Raspeberry Pi 4
ROS: ros-noetic-depthai-ros

I installed ros from bianary.

So I'm going to run rtabmap slam on ros1

But I'm getting this error
[image]

Embedded1993, 2023-03-28:
Yes. I have installed rtabmap-ros

Serafadam, 2023-03-28:
Could you paste what package version you have? 
`apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`

Embedded1993, 2023-03-28:
ubuntu@ubuntu:~$ apt show ros-noetic-depthai ros-noetic-depthai-ros ros-noetic-depthai-bridge ros-noetic-depthai-ros-msgs ros-noetic-depthai-ros-driver
Package: ros-noetic-depthai
Version: 2.20.2-1focal.20230202.200823
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 21.3 MB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.5), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-noetic-catkin
Homepage: https://www.luxonis.com/
Download-Size: 10.2 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-noetic-depthai-ros
Version: 2.6.1-2focal.20230225.012112
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 18.4 kB
Depends: ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-depthai-examples, ros-noetic-depthai-ros-driver, ros-noetic-depthai-ros-msgs
Download-Size: 2124 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: The depthai-ros package

Package: ros-noetic-depthai-bridge
Version: 2.6.1-2focal.20230216.065040
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 69.5 MB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libboost-dev, libopencv-dev, ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-ros-msgs, ros-noetic-image-transport, ros-noetic-robot-state-publisher, ros-noetic-ros-environment, ros-noetic-roscpp, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-stereo-msgs, ros-noetic-vision-msgs, ros-noetic-xacro
Download-Size: 10.5 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: The depthai_bridge package

Package: ros-noetic-depthai-ros-msgs
Version: 2.6.1-2focal.20230216.064856
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 475 kB
Depends: ros-noetic-geometry-msgs, ros-noetic-message-runtime, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-vision-msgs
Download-Size: 39.4 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-noetic-depthai-ros-driver
Version: 2.6.1-2focal.20230225.010515
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 1490 kB
Depends: libc6 (>= 2.17), libconsole-bridge0.4, libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 5.2), ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-diagnostic-msgs, ros-noetic-dynamic-reconfigure, ros-noetic-image-pipeline, ros-noetic-image-transport, ros-noetic-image-transport-plugins, ros-noetic-message-filters, ros-noetic-nodelet, ros-noetic-roscpp, ros-noetic-rospy, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-std-srvs, ros-noetic-vision-msgs
Download-Size: 277 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: Depthai ROS Monolithic node.

Embedded1993, 2023-03-28:
same issue when I run it.
roslaunch depthai_ros_driver camera.launch
But works this
roslaunch depthai_examples stereo_inertial_node.launch

Serafadam, 2023-03-28:
Ok, got it, it seems like newest version of packages wasn't added to latest noetic sync. If you don't want to wait, you could either build packages from source switch to testing repository to download latest packages.
To do that, edit `/etc/apt/sources.list.d/ros1-latest.list` and change `ros` to `ros-testing`

Embedded1993, 2023-03-28:
I can't find testing repository.
And I tried to build from source. But it crashed at 63% and frozen screen

Embedded1993, 2023-03-29:
> Ok, got it, it seems like newest version of packages wasn't added to latest noetic sync. If you don't want to wait, you could either build packages from source switch to testing repository to [...]

Please let me know

Embedded1993, 2023-03-29:
I built success from source. But I'm getting these warnings.

ubuntu@ubuntu:~/work/dai_ws$ roslaunch rtabmap_ros rgbd_mapping.launch
... logging to /home/ubuntu/.ros/log/b733bd48-ce38-11ed-9b40-9dd08eea8197/roslaunch-ubuntu-6737.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36687/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: /odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
[... 143 lines cut ...]
   /odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
[ INFO] [1680097868.712917447]: rtabmapviz started.
[ WARN] [1680097869.555629630]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
[ WARN] [1680097873.017379654]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info
[ WARN] [1680097873.713013828]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info

Embedded1993, 2023-03-29:
I have checked topic also.

ubuntu@ubuntu:~$ rostopic hz odom /camera/rgb/image_rect_color  /camera/depth_registered/image_raw /camera/rgb/camera_info /rtabmap/odom_info
subscribed to [/odom]
subscribed to [/camera/rgb/image_rect_color]
subscribed to [/camera/depth_registered/image_raw]
subscribed to [/camera/rgb/camera_info]
subscribed to [/rtabmap/odom_info]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages

Serafadam, 2023-03-30:
Hi, you should remap topics in rtabmap launch (see the example launch file). Also a minor note - if you have a camera with autofocus enabled, that might cause issues with odometry at startup, to solve that you can either set reset odometry via service/in rtabmap gui or set manual focus for the camera driver.

Embedded1993, 2023-03-30:
Thank you. how to remap topic in oak-d? 
Sorry my stupied question.

Embedded1993, 2023-03-31:
I'm getting this error when run roslaunch **depthai_examples stereo_inertial_node.launch**

[1844301011C2740E00] [1.1.2] [107.001] [system] [info] Cpu Usage - LeonOS 18.41%, LeonRT: 3.74%
[1844301011C2740E00] [1.1.2] [108.002] [system] [info] Memory Usage - DDR: 71.40 / 340.42 MiB, CMX: 2.09 / 2.50 MiB, LeonOS Heap: 36.16 / 77.32 MiB, LeonRT Heap: 4.13 / 41.23 MiB
[1844301011C2740E00] [1.1.2] [108.002] [system] [info] Temperatures - Average: 37.53 °C, CSS: 38.18 °C, MSS 36.77 °C, UPA: 37.71 °C, DSS: 37.48 °C
[1844301011C2740E00] [1.1.2] [108.002] [system] [info] Cpu Usage - LeonOS 19.67%, LeonRT: 4.30%
[2023-03-31 17:48:06.958] [warning] Monitor thread (device: 1844301011C2740E00 [1.1.2]) - **ping was missed, closing the device connection
[rviz-7] process has died [pid 51943, exit code -9, cmd /opt/ros/noetic/lib/rviz/rviz -d /home/ubuntu/work/dai_ws/src/depthai-ros/depthai_examples/rviz/stereoInertialROS1.rviz __name:=rviz __log:=/home/ubuntu/.ros/log/e35d4600-cfeb-11ed-9a8b-993cf8ad3baf/rviz-7.log].**
log file: /home/ubuntu/.ros/log/e35d4600-cfeb-11ed-9a8b-993cf8ad3baf/rviz-7*.log
[depth_image_to_pointcloud-6] process has finished cleanly
log file: /home/ubuntu/.ros/log/e35d4600-cfeb-11ed-9a8b-993cf8ad3baf/depth_image_to_pointcloud-6*.log
[depth_image_convertion_nodelet-5] process has finished cleanly
log file: /home/ubuntu/.ros/log/e35d4600-cfeb-11ed-9a8b-993cf8ad3baf/depth_image_convertion_nodelet-5*.log

I'm not sure why camera warn

[image]