luxonis/depthai-ros #266: [BUG] Camera stream delayed

etigafy, 2023-03-29:
I'm using the oak-d pro on the TB4 and running a pose estimation to drive towards a dock. I'm using the following parameters:
/oakd:
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: false
      i_floodlight_brightness: 0
      i_laser_dot_brightness: 100
      i_nn_type: none
      i_pipeline_type: RGB
      i_usb_speed: SUPER_PLUS
    rgb:
      i_board_socket_id: 0
      i_fps: 30.0
      i_height: 720
      i_interleaved: false
      i_max_q_size: 10
      i_preview_size: 250
      i_enable_preview: false
      i_low_bandwidth: true
      i_keep_preview_aspect_ratio: true
      i_publish_topic: true
      i_resolution: '1080'
      i_width: 1280
    use_sim_time: false
with the  1280x720 image size the image stream is delayed about 3s which is really bad for a docking scenario. The topic /oakd/rgb/image_raw/compressed is publishing with 29 Hz. 

I tried to use smaller image sizes but got troubles with the pose estimation which might be due to the camera calibration which i got from device.readCalibration() are these values not accurate enough for smaller images? 

Is there the possibility to just publish the compressed image just in monochrome?

Serafadam, 2023-03-30:
Hi, I assume this delay is occurring when streaming images over a network to some other device? If that's the case there's a lot of things that can influence latency, mostly network speed and setup. Some tips - you can try out [DDS Tuning](https://docs.ros.org/en/rolling/How-To-Guides/DDS-tuning.html) and using for example preview stream instead of RGB. Newest[ PR for Humble](https://github.com/luxonis/depthai-ros/pull/267) adds more ways you can also resize your RGB images.

etigafy, 2023-04-25:
Thank you for answering. Yes, exactly I'm running the Cam Nodes on a Raspberry PI as I'm using a Turtlebot4. Is there a way to resize images with the older version? As I don't want to mess up the Turtlebot4 image with updating the depthai-ros package. Is it also possible to get accurate camera calibration data for different image sizes? Is there a possibility to publish the images just in monochrome? Thanks in advance.

Serafadam, 2023-04-26:
Hi, other way to set custom size would be to build custom node as in depthai_examples. Updating driver shouldn't change anything significant, though you could also use Docker container to get latest camera driver in isolated environment. Camera info gets updated based on new size, and regarding monochrome, it's not available in the driver itself, although you could just make a simple node that subscribes and republishes those images, same way as in depthai_filters.

closed 2023-09-11 by Serafadam