luxonis/depthai-ros #268: [BUG] {Catkin_make_isolated unknown build_type}

ditodamaru, 2023-04-01:
**Describe the bug**
Hi. I am trying to use ORB3 using OAK-D and ROS Noetic based on this [procedure](https://qiita.com/nindanaoto/items/20858eca08aad90b5bab)

after I reached step Installing depthai-ros, and try to run catkin_make_isolated, I got this error :

    ubuntu@noetic:~/catkin_ws$ catkin_make_isolated
    Base path: /home/ubuntu/catkin_ws
    Source space: /home/ubuntu/catkin_ws/src
    Build space: /home/ubuntu/catkin_ws/build_isolated
    Devel space: /home/ubuntu/catkin_ws/devel_isolated
    Install space: /home/ubuntu/catkin_ws/install_isolated
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    ~~  traversing 7 packages in topological order:
    ~~  - depthai-ros (unknown)
    ~~  - depthai_descriptions (unknown)
    ~~  - depthai_filters (unknown)
    ~~  - depthai_ros_msgs (unknown)
    ~~  - depthai_bridge (unknown)
    ~~  - depthai_examples (unknown)
    ~~  - depthai_ros_driver (unknown)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    Error: Packages with unknown build types exist
    Can not build workspace with packages of unknown build_type

How to handle this error?

[image]

Detailed information : 
 - Ubuntu 20 and ROS Noetic running on LXD
 - Here is the output of ldconfig -p | grep depthai
 
        ubuntu@noetic:~/catkin_ws$ ldconfig -p | grep depthai
        libdepthai-opencv.so (libc6,x86-64) => /usr/local/lib/libdepthai-opencv.so
        libdepthai-core.so (libc6,x86-64) => /usr/local/lib/libdepthai-core.so

 Here is the ROS depthai package version I've installed :
 
        Package: ros-noetic-depthai
        Version: 2.20.2-1focal.20230202.195954
        Priority: optional
        Section: misc
        Maintainer: Sachin Guruswamy <sachin@luxonis.com>
        Installed-Size: 21.6 MB
        Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-noetic-catkin
        Homepage: https://www.luxonis.com/
        Download-Size: 10.3 MB
        APT-Manual-Installed: yes
        APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
        Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform
        
        Package: ros-noetic-depthai-ros
        Version: 2.6.1-2focal.20230225.005728
        Priority: optional
        Section: misc
        Maintainer: sachin <sachin@luxonis.com>
        Installed-Size: 18.4 kB
        Depends: ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-depthai-examples, ros-noetic-depthai-ros-driver, ros-noetic-depthai-ros-msgs
        Download-Size: 2132 B
        APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
        Description: The depthai-ros package
        
        Package: ros-noetic-depthai-bridge
        Version: 2.6.1-2focal.20230216.010713
        Priority: optional
        Section: misc
        Maintainer: Sachin Guruswamy <sachin@luxonis.com>
        Installed-Size: 69.5 MB
        Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libboost-dev, libopencv-dev, ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-ros-msgs, ros-noetic-image-transport, ros-noetic-robot-state-publisher, ros-noetic-ros-environment, ros-noetic-roscpp, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-stereo-msgs, ros-noetic-vision-msgs, ros-noetic-xacro
        Download-Size: 10.5 MB
        APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
        Description: The depthai_bridge package
        
        Package: ros-noetic-depthai-ros-msgs
        Version: 2.6.1-2focal.20230216.010505
        Priority: optional
        Section: misc
        Maintainer: Sachin Guruswamy <sachin@luxonis.com>
        Installed-Size: 475 kB
        Depends: ros-noetic-geometry-msgs, ros-noetic-message-runtime, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-vision-msgs
        Download-Size: 39.4 kB
        APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
        Description: Package to keep interface independent of the driver
        
        Package: ros-noetic-depthai-ros-driver
        Version: 2.6.1-2focal.20230225.004636
        Priority: optional
        Section: misc
        Maintainer: Adam Serafin <adam.serafin@luxonis.com>
        Installed-Size: 1609 kB
        Depends: libc6 (>= 2.14), libconsole-bridge0.4, libgcc-s1 (>= 3.0), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 5.2), ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-diagnostic-msgs, ros-noetic-dynamic-reconfigure, ros-noetic-image-pipeline, ros-noetic-image-transport, ros-noetic-image-transport-plugins, ros-noetic-message-filters, ros-noetic-nodelet, ros-noetic-roscpp, ros-noetic-rospy, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-std-srvs, ros-noetic-vision-msgs
        Download-Size: 315 kB
        APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
        Description: Depthai ROS Monolithic node.

Serafadam, 2023-04-03:
Hi, you should check out `noetic` branch, as default git branch is `humble`. Check out this repository's readme on building, also there are some new updates that might be handy

ditodamaru, 2023-04-04:
Hi. I got it. Thank you for your reply. I just started using GitHub and tried to post a question about my difficulties while implementing ROS and ORB SLAM. I put my question in the wrong place; it shouldn't be in the BUG report but just on "Something else". I will drop my question in a more proper context next time if I got a problem. Thank you

closed 2023-04-04 by ditodamaru