luxonis/depthai-ros #271: [BUG] Poor performance / Pi crashing when running ROS 2 MobilePublisher example

avielbr, 2023-04-01:
Hi everyone,

I have an OAK-D (not lite) camera connected to a Raspberry Pi 4 and I am trying to get the example launch files working according to the readme.

For some background:
- I have successfully run the Python MobileNet example [here](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/raspberrypi.html) over SSH with no problems. I'm getting good FPS and clear images.
- I am supplying the OAK-D with its own dedicated power source (a healthy 5V with 3A available).

I want to be able to use the ROS 2 launch files, so I followed the steps for installing the DepthAI ROS 2 binaries and I am able to run the launch files, for instance running `ros2 launch depthai_examples rgb_publisher.launch.py camera_model:=OAK-D`, and then checking (**from the Pi shell**) `ros2 topic hz /color/image` I see an average FPS of around 13.

So far so good!

But once I try to access the topic from my main laptop (the one connected to the Pi over SSH) I run into issues. Running the same `ros2 topic hz /color/image` 9 times out of 10 doesn't show anything, and when it does it's a very low FPS like 2.3. Running `ros2 topic echo /color/image` every once in a while I see an image published but most of the time I see:

```
A message was lost!!!
	total count change:27
	total count: 1150---
```

After leaving the `echo` running for a few seconds the Pi usually just crashes entirely.

Same behavior when I try and view the image stream from RViz2 -- sometimes I will manage to see one or two images from the `/color/image` topic, but very quickly the Pi simply crashes and I have to start over.

---

 - Running Ubuntu 22.04 Jammy desktop version (on both the Pi and main computer)
 - Using ROS 2 Humble
 - Installed DepthAI using ros binaries
 
Output of `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver `:

```
Package: ros-humble-depthai
Version: 2.20.2-1jammy.20230202.204256
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 20.1 MB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 4.5), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-humble-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 10.3 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-humble-depthai-ros
Version: 2.6.4-1jammy.20230303.010530
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 42.0 kB
Depends: ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-examples, ros-humble-depthai-ros-driver, ros-humble-depthai-ros-msgs, ros-humble-ros-workspace
Download-Size: 5,232 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: The depthai-ros package

Package: ros-humble-depthai-bridge
Version: 2.6.4-1jammy.20230302.223904
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 69.4 MB
Depends: libc6 (>= 2.32), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgcodecs4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libboost-dev, libopencv-dev, ros-humble-camera-info-manager, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-ros-msgs, ros-humble-image-transport, ros-humble-rclcpp, ros-humble-robot-state-publisher, ros-humble-ros-environment, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-stereo-msgs, ros-humble-vision-msgs, ros-humble-xacro, ros-humble-ros-workspace
Download-Size: 13.4 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.6.4-1jammy.20230302.210403
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1,310 kB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 106 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.6.4-1jammy.20230303.005120
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 844 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libopencv-stitching4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-diagnostic-msgs, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 213 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: Depthai ROS Monolithic node.
```

avielbr, 2023-04-01:
I realize this may be a deeper ROS 2 specific issue. I couldn't find my specific issue addressed in more general threads (non specific to OAK-D) but if anyone is aware of something of this sort please set me straight.

Serafadam, 2023-04-03:
Hi, I guess this is pretty similar to https://github.com/luxonis/depthai-ros/issues/266, outside of things I suggested there, I'd try using compressed topics instead of raw to view images.

avielbr, 2023-04-04:
Thanks @Serafadam, I'll try compressed images and report back. If I'm not mistaken compressed images aren't published by default -- can you help a n00b out with this? 😅