luxonis/depthai-ros #276: Rock Pi shutdown

mtgizaw15, 2023-04-06:
I have Ubuntu 20.04 and ROS 2 Foxy running on a Rock Pi 4 SE and I'm using an Oak-D-Lite camera. If I launch the rgb_stereo_node file(command: ros2 launch depthai_examples rgb_stereo_node.launch.py cameramodel:=OAK-D-LITE) it starts up normal I checked topics everything running good but I want to use RTAB-Map as well and if I start the RTAB-Map in a new terminal it runs good until a few minutes then the Rock Pi shuts down by itself. I get no errors and nothing. After this I tried to record some topics in a bag only with the rgb_stereo_node launched but as I start the recording after a few seconds the Rock Pi shuts down again.

My command to start the RTAB-Map:

ros2 launch rtabmap_ros rtabmap.launch.py \

rtabmap_args:="--delete_db_on_start" \

rgb_topic:=/color/video/image \

depth_topic:=/stereo/depth \

camera_info_topic:=/color/video/camera_info \

frame_id:=base_link \

approx_sync:=false \

wait_imu_to_init:=true \

qos:=1

I don't know if this is because of the Rock Pi or not. I appreciate any help even if someone know that if it's a Rock Pi issue.

Serafadam, 2023-04-06:
Hi, I don't have Rock Pi to test that setup, but I would start with analyzing RAM, CPU and temperature when running those examples

mtgizaw15, 2023-04-06:
Hi @Serafadam!
My first guess was the temperature as well. Maybe some powering issue is the root of the problem.
Thank you for your suggestion!

closed 2023-09-11 by Serafadam