luxonis/depthai-ros #278: Accel data not found. Dropping data from OAK-D (humble-v2.7.0)

Embedded1993, 2023-04-06:
**Check if issue already exists**
I check but I can't anything
- Google it (e.g. *Accel data not found. Dropping data* github luxonis depthai)

**Describe the bug**
I'm trying to run stereo_inertial_node.launch.py on Raspberry Pi 4(Ubuntu 22.04, Depthai-ros humble v2.7.0 )
But I'm getting these warning message.

[stereo_inertial_node-2] Device USB status: SUPER
[rviz2-4] [INFO] [1680761254.478996620] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1680761254.480716590] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-4] [INFO] [1680761260.239201318] [rviz2]: Stereo is NOT SUPPORTED
[component_container-3] [WARN] [1680761260.852396817] [point_cloud_xyzrgb_node]: New subscription discovered on topic '/stereo/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[rviz2-4] [INFO] [1680761269.411916689] [rviz2]: Stereo is NOT SUPPORTED
[stereo_inertial_node-2] [WARN] [1680761292.300578585] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761325.237649197] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761387.052965352] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761411.059457041] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761438.817423207] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761445.537712292] [IMU INTERPOLATION: ]: Accel data not found. Dropping data
[stereo_inertial_node-2] [WARN] [1680761456.934432074] [IMU INTERPOLATION: ]: Accel data not found. Dropping data

**Minimal Reproducible Example**
it works fine when run python3 depthai_demo.py

**Expected behavior**
want to see Accel and Gyro data

**Screenshots**
[image]

**Pipeline Graph**
[image]

**Attach system log**
 - Raspberry PI 4 - Ubuntu 22.04
 - ROS2 humble
 - build from humble-v2.7.0 source
 - I installed `depthai/depthai-core` library from rosdep
 - Please include versions of following packages - `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`
 - ubuntu@ubuntu:~/dai_ws$ apt show ros-humble-depthai ros-humble-depthai-ros ros-humble-depthai-bridge ros-humble-depthai-ros-msgs ros-humble-depthai-ros-driver
Package: ros-humble-depthai
Version: 2.20.2-1jammy.20230202.204256
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 20.1 MB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 4.5), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-humble-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 10.3 MB
APT-Manual-Installed: yes
APT-Sources: http://repo.ros2.org/ubuntu/main jammy/main arm64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-humble-depthai-ros
Version: 2.6.4-1jammy.20230303.010530
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 42.0 kB
Depends: ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-examples, ros-humble-depthai-ros-driver, ros-humble-depthai-ros-msgs, ros-humble-ros-workspace
Download-Size: 5232 B
APT-Sources: http://repo.ros2.org/ubuntu/main jammy/main arm64 Packages
Description: The depthai-ros package

Package: ros-humble-depthai-bridge
Version: 2.6.4-1jammy.20230302.223904
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 69.4 MB
Depends: libc6 (>= 2.32), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgcodecs4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libboost-dev, libopencv-dev, ros-humble-camera-info-manager, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-ros-msgs, ros-humble-image-transport, ros-humble-rclcpp, ros-humble-robot-state-publisher, ros-humble-ros-environment, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-stereo-msgs, ros-humble-vision-msgs, ros-humble-xacro, ros-humble-ros-workspace
Download-Size: 13.4 MB
APT-Sources: http://repo.ros2.org/ubuntu/main jammy/main arm64 Packages
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.6.4-1jammy.20230302.210403
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1310 kB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 106 kB
APT-Sources: http://repo.ros2.org/ubuntu/main jammy/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.6.4-1jammy.20230303.005120
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 844 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libopencv-stitching4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-diagnostic-msgs, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 213 kB
APT-Sources: http://repo.ros2.org/ubuntu/main jammy/main arm64 Packages
Description: Depthai ROS Monolithic node.

 - here is log files
 
 
[component_container_41778_1680763615275.log](https://github.com/luxonis/depthai-ros/files/11167207/component_container_41778_1680763615275.log)
[launch.log](https://github.com/luxonis/depthai-ros/files/11167208/launch.log)
[python3_41759_1680763614807.log](https://github.com/luxonis/depthai-ros/files/11167209/python3_41759_1680763614807.log)
[robot_state_publisher_41774_1680763615233.log](https://github.com/luxonis/depthai-ros/files/11167211/robot_state_publisher_41774_1680763615233.log)
[rviz2_41780_1680763616955.log](https://github.com/luxonis/depthai-ros/files/11167213/rviz2_41780_1680763616955.log)
[stereo_inertial_node_41776_1680763615645.log](https://github.com/luxonis/depthai-ros/files/11167214/stereo_inertial_node_41776_1680763615645.log)

saching13, 2023-04-06:
It's just dropping a gyro reading where accelerometer readings are not there for timestamp around it. You can Ignore it.

Embedded1993, 2023-04-06:
New subscription discovered on topic '/stereo/points', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY

Can I ignore it?

Embedded1993, 2023-04-06:
https://github.com/introlab/rtabmap/issues/742#issuecomment-1259965385

I'm trying to run it.

saching13, 2023-04-06:
There is a launch file for rtabmap in depthai_ros_driver. Try that.

Embedded1993, 2023-04-06:
I tried rtabmap in depthai_ros_driver. So I'm getting this warnings

[rgbd_odometry-1] [WARN] [1680765628.850452548] [rgbd_odometry]: The time difference between rgb and depth frames is high (diff=0.133347s, rgb=1680765627.328484s, depth=1680765627.461831s). You may want to set approx_sync_max_interval lower than 0.02s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.

sagarvl96, 2023-07-06:
Hello @Embedded1993! Did you resolve this issue? I looked into my tf tree and the problem is that the odom frame is always decoupled from the oak-d-base frame. I have no idea how to solve this. Do you have any idea? My tf tree:

[image]

Maybe @saching13 could suggest where the problem might be?

saching13, 2023-07-06:
^ @Serafadam

Serafadam, 2023-07-10:
Hi @sagarvl96, what's your exact setup?

sagarvl96, 2023-07-10:
Hello @Serafadam!

I am using Ubuntu 22.04.2 LTS. I am running it on the ROS2 Humble. I installed  the depthai_ros_driver package from rosdep. The versions of depthai packages I am using are as follows:

Package: ros-humble-depthai
Version: 2.22.0-1jammy.20230623.022440

Package: ros-humble-depthai-ros
Version: 2.7.3-1jammy.20230624.064620

Package: ros-humble-depthai-bridge
Version: 2.7.3-1jammy.20230623.090046

Package: ros-humble-depthai-ros-msgs
Version: 2.7.3-1jammy.20230623.055642

Package: ros-humble-depthai-ros-driver
Version: 2.7.3-1jammy.20230624.063735

I launched the rtabmap by running the following command: ros2 launch depthai_ros_driver rtabmap.launch.py (I have not made any changes to the parameters)

I am using the OAK-D-S2 USB camera. I have installed the camera on a robot and I am trying to map out the entire floor of a building.

Please let me know if I am missing any other information and I would be glad to share it with you!

tolas92, 2023-12-12:
hello i am also facing the same issue, any updates on this?

mandrievich, 2024-10-13:
Hi I have the same issue for ROS2 Jazzy version with no parameters changed.
Also if changed, this doesn't help: https://docs.luxonis.com/software/ros/depthai-ros/driver/#DepthAI%20ROS%20Driver-Syncing

Serafadam, 2024-10-14:
Hi, this issue can come from ROS2 network/DDS setup, you can try tuning DDS according to [this guide](https://docs.ros.org/en/jazzy/How-To-Guides/DDS-tuning.html), reducing camera resolution, using IPC transport can also help here.