luxonis/depthai-ros #280: [Question] Unable to launch camera.launch file in depthai_ros_driver package

cwillboyd, 2023-04-09:
I was told to post my questions about the depthai-ros package here rather than in the Luxonis Discord. 

I am trying to launch the camera.launch file in depthai_ros_driver package but I am getting the error " [FATAL] [1681073113.027777298]: Failed to unload nodelet '/oak` from manager `oak_nodelet_manager'". I am able to run the launch file with no errors when I follow the steps on the README and launch from the dai_ws folder in my home directory. However when I import the depthai_ros as a submodule into my ROS Noetic git repo and try to launch the same camera.launch file I an getting the error. Can anyone provide suggestions on how to troubleshoot this? Is this potentially an issue with the build environment or dependencies ? I have included the entire terminal output below. Thank You!

```
... logging to /home/christopher/.ros/log/abe6848e-d716-11ed-9541-65c17a58062e/roslaunch-cwb-ubuntu-9178.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cwb-ubuntu:36203/

SUMMARY
========

PARAMETERS
 * /oak/camera_i_enable_imu: True
 * /oak/camera_i_enable_ir: True
 * /oak/camera_i_floodlight_brightness: 0.0
 * /oak/camera_i_ip: 
 * /oak/camera_i_laser_dot_brightness: 800.0
 * /oak/camera_i_mx_id: 
 * /oak/camera_i_nn_type: spatial
 * /oak/camera_i_pipeline_type: RGBD
 * /oak/camera_i_usb_port_id: 
 * /oak/camera_i_usb_speed: SUPER_PLUS
 * /oak/imu_i_max_q_size: 30
 * /oak/left_i_board_socket_id: 1
 * /oak/left_i_calibration_file: 
 * /oak/left_i_fps: 30.0
 * /oak/left_i_height: 720
 * /oak/left_i_low_bandwidth: False
 * /oak/left_i_low_bandwidth_quality: 50
 * /oak/left_i_max_q_size: 30
 * /oak/left_i_preview_size: 416
 * /oak/left_i_publish_topic: False
 * /oak/left_i_resolution: 720
 * /oak/left_i_set_isp_scale: True
 * /oak/left_i_width: 1280
 * /oak/left_r_exposure: 1000
 * /oak/left_r_focus: 1
 * /oak/left_r_iso: 800
 * /oak/left_r_keep_preview_aspect_ratio: True
 * /oak/left_r_set_man_exposure: False
 * /oak/left_r_set_man_focus: False
 * /oak/left_r_set_man_whitebalance: False
 * /oak/left_r_whitebalance: 3300
 * /oak/nn_i_nn_config_path: depthai_ros_drive...
 * /oak/rgb/image_raw/compressed/format: jpeg
 * /oak/rgb/image_raw/compressed/jpeg_optimize: False
 * /oak/rgb/image_raw/compressed/jpeg_progressive: False
 * /oak/rgb/image_raw/compressed/jpeg_quality: 80
 * /oak/rgb/image_raw/compressed/jpeg_restart_interval: 0
 * /oak/rgb/image_raw/compressed/png_level: 9
 * /oak/rgb/image_raw/compressedDepth/depth_max: 10.0
 * /oak/rgb/image_raw/compressedDepth/depth_quantization: 100.0
 * /oak/rgb/image_raw/compressedDepth/format: png
 * /oak/rgb/image_raw/compressedDepth/png_level: 9
 * /oak/rgb/image_raw/theora/keyframe_frequency: 64
 * /oak/rgb/image_raw/theora/optimize_for: 1
 * /oak/rgb/image_raw/theora/quality: 31
 * /oak/rgb/image_raw/theora/target_bitrate: 800000
 * /oak/rgb_i_board_socket_id: 0
 * /oak/rgb_i_calibration_file: 
 * /oak/rgb_i_enable_preview: False
 * /oak/rgb_i_fps: 60.0
 * /oak/rgb_i_height: 720
 * /oak/rgb_i_interleaved: True
 * /oak/rgb_i_keep_preview_aspect_ratio: True
 * /oak/rgb_i_low_bandwidth: False
[... 113 lines cut ...]
[ INFO] [1681072799.086022067]: Camera ready!
[ERROR] [1681073112.879800834]: Cannot load nodelet /oak for one exists with that name already
[ WARN] [1681073112.926655114]: Shutdown request received.
[ WARN] [1681073112.930997726]: Reason given for shutdown: [[/oak_state_publisher] Reason: new node registered with same name]
[ WARN] [1681073112.936363754]: Shutdown request received. Reason: [[/oak] Reason: new node registered with same name]
[ WARN] [1681073112.941062096]: Shutdown request received.
[ WARN] [1681073112.941212669]: Reason given for shutdown: [[/oak_nodelet_manager] Reason: new node registered with same name]
[ WARN] [1681073112.947915743]: Shutdown request received.
[ WARN] [1681073112.947988925]: Reason given for shutdown: [[/oak_state_publisher] Reason: new node registered with same name]
[ INFO] [1681073113.025333827]: Unloading nodelet /oak from manager oak_nodelet_manager
[FATAL] [1681073113.027777298]: Failed to unload nodelet '/oak` from manager `oak_nodelet_manager'
================================================================================REQUIRED process [oak_state_publisher-2] has died!
process has finished cleanly
log file: /home/christopher/.ros/log/abe6848e-d716-11ed-9541-65c17a58062e/oak_state_publisher-2*.log
Initiating shutdown!
================================================================================
[oak-4] killing on exit
[oak_nodelet_manager-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Serafadam, 2023-04-12:
Hi, based on 
`[ WARN] [1681073112.947921975]: Reason given for shutdown: [[/oak_state_publisher] Reason: new node registered with same name]`
it seems that you launch state publisher node twice, could you show how your launch file looks?

cwillboyd, 2023-04-12:
I was able to fix the issue by removing the depthai-ros submodule, reinstalling the depthai-ros submodule, and running `sudo apt install ros-noetic-depthai-ros-*`

closed 2023-04-12 by cwillboyd